todbot/blink1 #197: bug in blink1 or kernel driver (not sure which)

frioux, 2015-01-06:
I just noticed though that there's a fairly frustrating bug, though I'm not totally sure where it lies or how to trigger it.  Basically, if you get an error writing a color to your blink, it logs to the kernel log, viewable by dmesg, but it logs some terminal escapes and trashes the console.   Here is my example:

      [1227993.036354] hid-generic 0003:1050:0010.001C: input,hidraw5: USB HID v1.11 Keyboard [Yubico Yubico Yubikey II] on usb-0000:00:14.0-2.4.2/input0
      [1232211.129647] hid \xffffffc0\xffffff82S\│°°°°°°88\│°°°°°°°°\│°°°°°°°\│°°°°°°89S\│°°°°°°88\│°°°°°°°°\│°°°°°°°\│°°°°°°89S\│°°°°°°88\│°°°°°°°°\│°°°°°°°°: °▒☃┌ed ├⎺ ┬⎼☃├e c⎺┌⎺⎼
      [1232211↓129765] ▒☃d \│°°°°°°c▮\│°°°°°°82S\│°°°°°°88\│°°°°°°°°\│°°°°°°°\│°°°°°°89S\│°°°°°°88\│°°°°°°°°\│°°°°°°°\│°°°°°°89S\│°°°°°°88\│°°°°°°°°\│°°°°°°°°: °▒☃┌ed ├⎺ ┬⎼☃├e c⎺┌⎺⎼

If we look at it with xxd it becomes clear that it's a blink issue:

      001a620: 3430 0a5b 3132 3237 3939 332e 3033 3633  40.[1227993.0363
      001a630: 3534 5d20 6869 642d 6765 6e65 7269 6320  54] hid-generic 
      001a640: 3030 3033 3a31 3035 303a 3030 3130 2e30  0003:1050:0010.0
      001a650: 3031 433a 2069 6e70 7574 2c68 6964 7261  01C: input,hidra
      001a660: 7735 3a20 5553 4220 4849 4420 7631 2e31  w5: USB HID v1.1
      001a670: 3120 4b65 7962 6f61 7264 205b 5975 6269  1 Keyboard [Yubi
      001a680: 636f 2059 7562 6963 6f20 5975 6269 6b65  co Yubico Yubike
      001a690: 7920 4949 5d20 6f6e 2075 7362 2d30 3030  y II] on usb-000
      001a6a0: 303a 3030 3a31 342e 302d 322e 342e 322f  0:00:14.0-2.4.2/
      001a6b0: 696e 7075 7430 0a5b 3132 3332 3231 312e  input0.[1232211.
      001a6c0: 3132 3936 3437 5d20 6869 6420 5c78 6666  129647] hid \xff
      001a6d0: 6666 6666 6330 5c78 6666 6666 6666 3832  ffffc0\xffffff82
      001a6e0: 530e 015c 7866 6666 6666 6638 385c 7866  S..\xffffff88\xf
      001a6f0: 6666 6666 6666 665c 7866 6666 6666 6666  fffffff\xfffffff
      001a700: 6608 5c78 6666 6666 6666 3839 530e 015c  f.\xffffff89S..\
      001a710: 7866 6666 6666 6638 385c 7866 6666 6666  xffffff88\xfffff
      001a720: 6666 665c 7866 6666 6666 6666 6608 5c78  fff\xffffffff.\x
      001a730: 6666 6666 6666 3839 530e 015c 7866 6666  ffffff89S..\xfff
      001a740: 6666 6638 385c 7866 6666 6666 6666 665c  fff88\xffffffff\
      001a750: 7866 6666 6666 6666 663a 2066 6169 6c65  xffffffff: faile
      001a760: 6420 746f 2077 7269 7465 2063 6f6c 6f72  d to write color
      001a770: 0a5b 3132 3332 3231 312e 3132 3937 3635  .[1232211.129765
      001a780: 5d20 6869 6420 5c78 6666 6666 6666 6330  ] hid \xffffffc0
      001a790: 5c78 6666 6666 6666 3832 530e 015c 7866  \xffffff82S..\xf
      001a7a0: 6666 6666 6638 385c 7866 6666 6666 6666  fffff88\xfffffff
      001a7b0: 665c 7866 6666 6666 6666 6608 5c78 6666  f\xffffffff.\xff
      001a7c0: 6666 6666 3839 530e 015c 7866 6666 6666  ffff89S..\xfffff
      001a7d0: 6638 385c 7866 6666 6666 6666 665c 7866  f88\xffffffff\xf
      001a7e0: 6666 6666 6666 6608 5c78 6666 6666 6666  fffffff.\xffffff
      001a7f0: 3839 530e 015c 7866 6666 6666 6638 385c  89S..\xffffff88\
      001a800: 7866 6666 6666 6666 665c 7866 6666 6666  xffffffff\xfffff
      001a810: 6666 663a 2066 6169 6c65 6420 746f 2077  fff: failed to w
      001a820: 7269 7465 2063 6f6c 6f72 0a              rite color.

frioux, 2015-01-06:
Oh and for what it's worth, my firmware version is 203, not sure what the version of blink1-tool is though, as it apparently doesn't have one?

frioux, 2015-01-06:
Blahhh, and when I unplugged and then replugged my blink to try to repro this, my whole computer crashed.

frioux, 2015-01-06:
After the aformentioned forced reboot, I now see this in the dmesg logs:

      [   75.415758] hid `Dn\xffffff88\xffffffff\xffffffff8:01ED.0007: failed to write color
      [   75.415902] hid `Dn\xffffff88\xffffffff\xffffffff8:01ED.0007: failed to write color
      [   75.416016] hid `Dn\xffffff88\xffffffff\xffffffff8:01ED.0007: failed to write color

So I guess a kernel update may have fixed that, but my blink still has a cray cray name?

todbot, 2015-01-09:
Hi! 
What is the exact distro and Linux version you are using?
What are you exact command-line commands you are using to control the blink(1)?
Are you using a pre-compiled blink1-tool, or something else?
If you are using a pre-compiled binary, what is the download URL of that binary?

frioux, 2015-01-09:
Ubuntu 14.10, kernel at this moment is Linux gae-bulg 3.16.0-28-generic #38-Ubuntu SMP Fri Dec 12 17:37:40 UTC 2014 x86_64 x86_64 x86_64 GNU/Linux, didn't think to check at the time.

I got the blink1-tool from this github account under releases: https://github.com/todbot/blink1/releases/download/v1.93/blink1-tool-v1.93-linux-x86_64.zip

todbot, 2015-01-09:
Thanks. Are you using any sort of special USB kernel module or anything non-standard in your distro?  I've not seen this message before and it looks like it's pretty low in the Linux USB HID stack (and not something we can control with user-level programs like blink1-tool)

It may be a problem between libusb versions between the system we built the binary on and your system.  So one thing to try if you have time: do a checkout of this repo and build a version of blink1-tool on your system.  The short description of how to do this if you're familiar with building code by hand is:
```
# apt-get install build-essential pkg-config libusb-1.0-0-dev
# git clone https://github.com/todbot/blink1/
# cd blink1/commandline
# make
# ./blink1-tool --on
```

frioux, 2015-01-09:
ok, I'll try it when I get a minute; at one point I think I tried building the blink1-tool and had issues, but I will try again.

apple4ever, 2015-02-17:
I've having the same issue on the same distro. I built blink1-tool from the repo, but it has the same issue.

Oddly, this does not show up when I run it manually. I created a bash script (so I can use it via Pidgin pounces), and it will randomly say "no blink(1) devices found" and the above "failed to write color" will show up in dmesg.

apple4ever, 2015-04-03:
I found I had to add a sleep .1 in my script and once I did that my scripts work.