todbot/blink1 #203: --blink option for blink1-tool on command line doesn't work as expected when trying to light multiple devices

ddd1600, 2015-01-23:
I've nearly pulled my hair out over this thing but have finally got it figured out. My solution is a hack, in some respects, so I wanted to let you guys know what's wrong.

So when I run this command

    blink1-tool --id all --on

All the lights cut on as expected

Same goes for:

    blink1-tool --id all --rgb=255,0,0

However, when I try to run this command:

    blink1-tool --id all --rgb=255,0,0 --blink=5

Only one light (the first one) goes off.

If I rearrange the order of these options like so:

    blink1-tool --blink=5 --rgb=255,0,0 --id all

Both lights glow but no blinking occurs.

Just wanted y'all to know this --blink thing is a defective function when running multiple devices.

ddd1600, 2015-01-23:
Anyway, so I've just broken out the functionality in Ruby and it all works fine. Took me a long time to figure this out though. Very frustrating.

todbot, 2015-01-24:
Hi,
You are right, the "--blink" command does not respond correctly with the "--id" option.  Apologies.  I'm working on a fix now.  Which OS are you on?

ddd1600, 2015-01-26:
osx mavericks

todbot, 2015-02-03:
Hi,
I've updated blink1-tool so that the "--blink" option works as expected. You can download it from the releases page:
  https://github.com/todbot/blink1/releases

There's also now some blink1-tool tips here: 
  https://github.com/todbot/blink1/blob/master/docs/blink1-tool-tips.md
(which will eventually become a proper webpage on the main site)

closed 2015-02-03 by todbot