todbot/blink1 #204: Blink1Control seems to crash

andreasloe, 2015-01-24:
I am starting Blink1Control from my Mac OS X (10.10.1) with different options and also after several restarts. But I cannot open IFTTT, Mail or any other page. Also, the Blink1-picture in the open left corner stays dark and does not have the colour I have choosen right below.

The strange thing is that if I select a color right below my stick shows me exactly that colour.

todbot, 2015-01-24:
Can you provide a screenshot of some of the issues your'e seeing?
When you say "starting it with different options" what do you mean?
What download link are you using to get Blink1Control?
Thanks!

andreasloe, 2015-01-24:
Screenshots: I see the "normal" window as in here
[image]
and all what can be seen (like IFTTT, Tools, Mail, Hardware, Help or Color Picker, Color Cycle etc.) can be clicked on and nothing happens.

I startet the software with "start minimized" on | off, "start at login" on | off, "enabled API" on | off.

The download link was https://github.com/todbot/blink1/releases/download/v1.94/Blink1Control-mac-install.dmg

todbot, 2015-01-24:
Thanks, I'm looking into it now.

todbot, 2015-01-24:
Okay, did some investigating.  There's currently a bug with either the "Start Minimized" and/or the "Start at Login" options.  Selecting them triggers the issue you're seeing.

A work-around for now is to launch Blink1Control, ctrl-click on its Dock icon, and deselect those two options.  Then quit Blink1Control (you may need to use Cmd-Q) and relaunch.

andreasloe, 2015-01-24:
Thanks, that worked!

todbot, 2015-10-01:
Hi,

We’ve recently released an updated version of Blink1Control, v1.98.
Hopefully it will address the issues you’ve been experiencing.

You can download the new version here:
http://blink1.thingm.com/blink1control/
Or on this repo's releases page

closed 2015-10-01 by todbot