todbot/blink1 #205: Having an issue building on Fedora Server 21

evanextreme, 2015-01-27:
I tried using the make command in the command line folder on a VM running Fedora Server 21. I am testing out multiple server environments and found that this was the best one for me, but I am having a problem with the makefile on it. Sorry if I am a bit of a noob at this, this is my first time working with this type of software.

```
Building for OS=linux BLINK1_VERSION=0-linux-x86_64
cc -DUSE_HIDAPI -I./hidapi/hidapi  `pkg-config libusb-1.0 --cflags` -fPIC -std=gnu99  -g -DBLINK1_VERSION=\"""0"-linux-"x86_64""\" -c hidapi/libusb/hid.c -o hidapi/libusb/hid.o
Package libusb-1.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `libusb-1.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libusb-1.0' found
hidapi/libusb/hid.c:47:20: fatal error: libusb.h: No such file or directory
```

todbot, 2015-01-27:
All the Linux distros are a bit different, so don't feel like you're a noob.  I try to put the right tips in the Makefile for particular systems, including Fedora, but I sometimes leave stuff out.

It looks like you need to install the libusb devel package. Try:
```
sudo yum install libusbx-devel
```
I dumbly just did "yum update" and am waiting for that to finish before I try the above.

evanextreme, 2015-01-27:
Thanks for the help! While it didn't fix the issues entirely, the message has changed so I feel that was one of the issues. When I tried to make again this is what I got
'''
cc -DUSE_HIDAPI -I./hidapi/hidapi  `pkg-config libusb-1.0 --cflags` -fPIC -std=gnu99  -g -DBLINK1_VERSION=\"""0"-linux-"x86_64""\" -c hidapi/libusb/hid.c -o hidapi/libusb/hid.o
cc -DUSE_HIDAPI -I./hidapi/hidapi  `pkg-config libusb-1.0 --cflags` -fPIC -std=gnu99  -g -DBLINK1_VERSION=\"""0"-linux-"x86_64""\" -c blink1-lib.c -o blink1-lib.o
cc -DUSE_HIDAPI -I./hidapi/hidapi  `pkg-config libusb-1.0 --cflags` -fPIC -std=gnu99  -g -DBLINK1_VERSION=\"""0"-linux-"x86_64""\"   -c -o blink1-tool.o blink1-tool.c
cc -DUSE_HIDAPI -I./hidapi/hidapi  `pkg-config libusb-1.0 --cflags` -fPIC -std=gnu99  -g -DBLINK1_VERSION=\"""0"-linux-"x86_64""\" -c blink1-tool.c -o blink1-tool.o
cc -DUSE_HIDAPI -I./hidapi/hidapi  `pkg-config libusb-1.0 --cflags` -fPIC -std=gnu99  -g -DBLINK1_VERSION=\"""0"-linux-"x86_64""\" -static -g ./hidapi/libusb/hid.o blink1-lib.o  `pkg-config libusb-1.0 --libs` -lrt -lpthread -ldl blink1-tool.o -o blink1-tool 
/usr/bin/ld: cannot find -lusb-1.0
/usr/bin/ld: cannot find -lrt
/usr/bin/ld: cannot find -lpthread
/usr/bin/ld: cannot find -ldl
/usr/bin/ld: cannot find -lc
collect2: error: ld returned 1 exit status
Makefile:343: recipe for target 'blink1-tool' failed
make: *** [blink1-tool] Error 1'''

todbot, 2015-01-27:
How did you get the source package?  Is it not from a recent git pull?

evanextreme, 2015-01-27:
It was, I cloned it today.

evanextreme, 2015-01-27:
I just did a pull again and it said it was already up to date

todbot, 2015-01-27:
You must be on an old branch or something because those build messages are from an old style of compilation.  What do you get when you do a `git branch`?

todbot, 2015-01-27:
Or perhaps you are pulling from a different remote than this one?  What do you see when you do `git remote -v`?

evanextreme, 2015-01-27:
Welp, I figured it out. For some reason I decided to pull from thingm/blink1 instead of tidbit/blink1. Thanks for the help even though it was a simple issue :)

closed 2015-01-27 by evanextreme

todbot, 2015-01-27:
Oh my bad for not updating thingm repo.  thanks for pointing that out.  Updating it now.

evanextreme, 2015-01-28:
No problem! Thanks for the help. With the magic of SUDO I managed to get everything working :+1: