todbot/blink1 #206: Compiler error when building mk2 firmware

Juggle4Evr, 2015-01-28:
When trying to compile the blink(1)mk2 firmware (from the "hardware/firmware_mk2" directory from GitHub), I get the following compiler error:

<i>usb_device.c:310: error: could not find space (1 byte) for variable _USBStatusStageTimeoutCounter</i>

I use MPLAB X IDE v2.00 (Windows), installed the "Microchip libraries for applications v2013-06-15".

It seems as if the compiler thinks the 16F1455 has not enough RAM. Got the same problem compiling on MacOS.

Any ideas?

todbot, 2015-01-28:
Hmm, never seen that error.  It might be because blink1mk2 firmware must be built with the PRO version of the compiler since it needs optimization to fit in the space available.  If you have the PRO version of the compiler, you'll probably need to select it by Project Properties / XC8 compiler / Optimizations / Operation Mode: PRO.

Juggle4Evr, 2015-01-28:
But doesn't the optimization only affect the code size but not the data size? The 16F1455 has 1k RAM, that sound quite a lot.
Unfotunatyle I don't have the PRO version.

todbot, 2015-01-28:
PICs are really weird, and code & memory can get duplicated if there are interrupts involved. I think the optimizations in XC8 help reduce this duplication.

Juggle4Evr, 2015-01-28:
Hm, ok, maybe there's a free limited time key for the PRO version. Or I look for a compatible PIC with more RAM (any suggestions?).

Juggle4Evr, 2015-01-28:
Ok, got the 60 day PRO evaluation: Compiles perfectly now. It uses 53% of the RAM of the PIC.
Seems the free mode is very generous with RAM consumption. Problem is, for hobby I just can't afford to buy the PRO version. Maybe I have to try to optimize RAM myself to squeeze out some more bytes.
Thanks anyway!

Juggle4Evr, 2015-01-28:
Sorry, the Dashboard says it used 53% (859 bytes) of RAM of a total of 1636 bytes. But the memory summary when compiling says it used 84% (859 bytes) of RAM of a total of 1024 bytes. It's a bit confsuing but I think the 83% are correct.

todbot, 2015-01-28:
The confusion over code/RAM percentage used, the PRO vs. FREE compiler, and many other things really bug me about PIC development.  Unfortunately, the PIC16F1455 is the only chip that could meet the needs of blink(1) at the time (and still really).  It's a pretty good chip with a frustrating compiler.