todbot/blink1 #207: blink1-tool v1.96 won't --blink

digitalcardboard, 2015-02-09:
At least in Windows, using the `--blink` option with blink1-tool doesn't do anything. Appears to work fine in v1.93.

(It's unfortunately difficult to search for this in the existing issues list, as the feature (blinking) happens to be the name of the product)

todbot, 2015-02-09:
Thanks. I just verified this on Win7. My quick unit tests didn't do a `--blink`.
Looks to be a problem with windows version of `sleep()`.  

Also, I'll add `--flash` as an alias to `--blink` in new builds for disambiguation purposes. :-)

todbot, 2015-02-09:
Nope, it was a problem with new helper function in `blink1-tool.c`.  It will be fixed for v1.97, to be released later today.  Sorry about that!

digitalcardboard, 2015-02-09:
Thanks and thanks! For now v1.93 is working for my needs, but it's great to see continued development.

digitalcardboard, 2015-02-09:
Works like a champ with `--blink/--flash`

closed 2015-02-09 by digitalcardboard