todbot/blink1 #208: Unable to startup Blink1Control on OSX 10.7.5

brucemcbruce, 2015-02-12:
I'm trying to run Blink1Control on OSX 10.7.5, running on an older-style mac mini (the 2007 model), and when I try to run Blink1Control I immediately get a "Blink1Control quit unexpectedly" error, and it doesn't open.

I am able to control the blink1 via the command line app, but I'm wanting to use the web API instead of having to do everything locally.

The diagnostic report is pasted below:

Process:         launchd [720]
Path:            /Applications/Blink1Control.app/Contents/MacOS/Blink1Control
Identifier:      com.thingm.Blink1Control
Version:         ??? (???)
Code Type:       X86-64 (Native)
Parent Process:  launchd [339]

Date/Time:       2015-02-12 10:57:04.411 +1100
OS Version:      Mac OS X 10.7.5 (11G63)
Report Version:  9

Interval Since Last Report:          15207166 sec
Crashes Since Last Report:           9
Per-App Crashes Since Last Report:   9
Anonymous UUID:                      D229EC6D-34A4-4BB6-9CA7-81EFD0EFE1DC

Crashed Thread:  Unknown

Exception Type:  EXC_BAD_ACCESS (SIGSEGV)
Exception Codes: KERN_INVALID_ADDRESS at 0x00007fff5fc01028

Backtrace not available

Unknown thread crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000055  rbx: 0x0000000000000000  rcx: 0x0000000000000000  rdx: 0x0000000000000000
  rdi: 0x0000000000000000  rsi: 0x0000000000000000  rbp: 0x0000000000000000  rsp: 0x0000000000000000
   r8: 0x0000000000000000   r9: 0x0000000000000000  r10: 0x0000000000000000  r11: 0x0000000000000000
  r12: 0x0000000000000000  r13: 0x0000000000000000  r14: 0x0000000000000000  r15: 0x0000000000000000
  rip: 0x00007fff5fc01028  rfl: 0x0000000000010203  cr2: 0x00007fff5fc01028
Logical CPU: 0

Binary images description not available

External Modification Summary:
  Calls made by other processes targeting this process:
    task_for_pid: 1
    thread_create: 0
    thread_set_state: 0
  Calls made by this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by all processes on this machine:
    task_for_pid: 567
    thread_create: 0
    thread_set_state: 0

Model: Macmini2,1, BootROM MM21.009A.B00, 2 processors, Intel Core 2 Duo, 1.83 GHz, 2 GB, SMC 1.19f2
Graphics: Intel GMA 950, GMA 950, Built-In, spdisplays_integrated_vram
Memory Module: BANK 0/DIMM0, 1 GB, DDR2 SDRAM, 667 MHz, 0xCE00000000000000, 0x4D342037305432393533455A332D43453620
Memory Module: BANK 1/DIMM1, 1 GB, DDR2 SDRAM, 667 MHz, 0xCE00000000000000, 0x4D342037305432393533455A332D43453620
AirPort: spairport_wireless_card_type_airport_extreme (0x168C, 0x86), Atheros 5424: 2.1.14.9
Bluetooth: Version 4.0.8f17, 2 service, 18 devices, 1 incoming serial ports
Network Service: Ethernet, Ethernet, en0
Serial ATA Device: WDC WD5000BEKT-00KA9T0, 500.11 GB
Parallel ATA Device: MATSHITACD-RW  CW-8124
USB Device: blink(1), 0x27b8, 0x01ed, 0x3d100000 / 2
USB Device: Bluetooth USB Host Controller, apple_vendor_id, 0x8205, 0x7d100000 / 2
USB Device: IR Receiver, apple_vendor_id, 0x8240, 0x7d200000 / 3

todbot, 2015-02-16:
Hi Bruce,
The current version of Blink1Control does not support Mac OS X 10.7. Apologies for that. The underlying technology (Qt) we use for Blink1Control has a huge number of issues on 10.7 that we weren't able to resolve.

You can try the previous incarnation of Blink1Control:  http://thingm.com/blink1/downloads/Blink1Control-0.997-applescript.zip

brucemcbruce, 2015-02-16:
Fair enough, that's what I get for using old hardware, especially when Apple helpfully won't let me update the OS :( I'll give the previous versions a shot.

Although if that's the case it's probably worth updating the download link at http://blink1.thingm.com/blink1control/ - it currently lists the installer as "Blink1Control for Mac OS X 10.7+".

closed 2015-02-16 by brucemcbruce

todbot, 2015-02-16:
oh you're right about the download link. doh! will fix.