todbot/blink1 #209: blink1-control force off

ftbastler, 2015-02-16:
Using a command like `sudo cmd/blink1-tool --rgb=255,0,00 --blink 100` works and will make the blink1 blink in red.

Now when you try to use a command like `sudo cmd/blink1-tool --off` the blink1 doesn't get shut off.

todbot, 2015-02-16:
Hi,
I'm unclear what you mean.  Are you trying to run `blink1-tool --off` while `blink1-tool --blink 100` is still running?  If so, you'll need to kill that process first to release its hold on the blink(1) before you can run `--off`

ftbastler, 2015-02-17:
Yes, that's what I am trying to do. How would I do that?

todbot, 2015-02-17:
On Mac OS X and most Linuxes, you can use the "pkill" command.  So if in one terminal window you did `blink1-tool --blink 100` then in another you could do `pkill blink1-tool && blink1-tool --off`

closed 2015-02-18 by ftbastler