todbot/blink1 #213: Blink1Control: Freezing During Launch & Random Crashing in Yosimite

BzowK, 2015-03-03:
Hey Guys - 

Just installed 1.94 (newest) Blink1Control on my MacBook Pro running Yosemite 10.10.2.  It was a fresh install as hasn't ever run any version of Blink1Control before.  Upon startup, it loads the window fast, but I cannot change tabs until about 20-30 seconds after I launch it.  This occurs regardless of if Blink1 is plugged in or not.  When it finally starts to work, I have tried creating patterns from copies of samples but one I get to starting on the 2nd one, the app crashes.

Environment
- MacBook Pro Retina Mid-2014
- Yosemite 10.10.2
- Do have Parallels 10 installed and Basecamp VM powered on which may affect USB communication
- Only changes from default install I made in Blink1Control was settings IFTTT key in my channel and sending test (which worked).  Also set it to start minimized

What I've Tried 
(None of the below change what's occurring in the slightest)
- Reboot OS
- Plugging Blink1 into other USB port
- Launching with & without Blink1 plugged in
- Initially downloaded DMG from website.  Uninstalled it (deleted .app) after having issues then went to GitHub page and downloaded 90mb zip.  Uncompressed it to .app, moved to Applications, ran it, and still had same issues.  Settings I made in first install remained but haven't yet found ini file to delete (move of a Windows guy :))

Any workarounds currently known?  Hope this helps - Thanks

todbot, 2015-03-03:
Hmm, that sounds frustrating.   Could it be related to issue #204?

And if you want to poke around, the config file is located in `~/.config/ThingM/Blink1Control.ini`

todbot, 2015-03-03:
Also, I'm doing some rebuilds using an updated Qt install.  Should have an alternative binary for you to try later today.

BzowK, 2015-03-03:
Thanks for your fast replies…  I didn’t see #204 earlier, but just reviewed it from the link you sent and yes, that seems to be the exact same issue.  I say this because the Blink1 picture in the upper left of mine is gray also even though showing it’s connected plus I’m starting minimized.  I don’t know if the icon (which I didn’t notice earlier), it talking a long to to become responsive after starting, and crashing are related; but hopefully so for your sake.  I’ll be happy to test out any builds if given a link or whatever else needed.  Might be able to test better if I had the newer Blink1 model, though.  :)

Thanks again for looking into it!

Ben Kellermann
ben@thekellermanns.net<mailto:ben@thekellermanns.net>
615.585.4518 (Mobile)

todbot, 2017-10-17:
This should be fixed in Blink1Control2 : https://github.com/todbot/Blink1Control2/releases

closed 2017-10-17 by todbot