todbot/blink1 #214: --blink 9999 only blinks 15 times

danpritts, 2015-03-04:
when i --blink 9999 i get only 15 blinks.... 

osx 10.10.2, macbook pro retina, blink(1) mk2 from kickstarter.

blink1-tool version: v1.97-macosx-x86_64, fw version: 203

% /Users/danno/bin/blink1-tool --red --blink 9999
blink 15 times rgb:ff,0,0:
set dev:0 to rgb:0xff,0x00,0x00 over 300 msec
set dev:0 to rgb:0x00,0x00,0x00 over 300 msec
[...]

Upon experimentation, It appears that the real problem is that the blink counter is an 8-bit integer.  

seriously?

~@precious% blink1-tool --blink 255
blink 255 times rgb:ff,ff,ff:
set dev:0 to rgb:0xff,0xff,0xff over 300 msec
set dev:0 to rgb:0x00,0x00,0x00 over 300 msec
^C
~@precious% blink1-tool --blink 256
blink 0 times rgb:ff,ff,ff:
~@precious% blink1-tool --blink 257
blink 1 times rgb:ff,ff,ff:
set dev:0 to rgb:0xff,0xff,0xff over 300 msec
set dev:0 to rgb:0x00,0x00,0x00 over 300 msec

todbot, 2015-03-04:
ahah oops.  That's awesomely dumb of me.  I'll make a change in the source now. New binaries up by this weekend.  It's because I was using the hex tuple arg parser instead of straight up strtol.

danpritts, 2015-03-04:
hey, at least you used an unsigned int. :)

I grabbed the source, and it works. thanks!

closed 2015-03-04 by danpritts