todbot/blink1 #217: Running as non-root on linux

peterfpeterson, 2015-03-16:
The `linux/README.md` has a mention of installing the `51-blink1.rules` which are needed to run as a user on linux. This file needs to be installed when installing `commandline/blink1-tool` from source. I can generate a pull request, but I don't know if moving files around is something that you want an outside contributor to do.

todbot, 2015-03-16:
Hi,
I'm not sure what your question is.  Are you saying to want the blink1-tool Makefile to include an install step for installing the udev rules?  I think that can be done.  In the past I've tried to make install-level issues not part of the blink1-tool Makefile because blink1-tool is so cross-platform.  But I think we can make it an install target and use good-enough platform detection. :)

peterfpeterson, 2015-03-16:
My naive approach was to go into `commandline` then `make && sudo make install` which works except for the missing rules. Maybe a better approach is mentioning it in the  `README.md` would be a better option.

todbot, 2015-03-16:
I think your approach is perfectly reasonable.  If you want to submit a pull-request with the changes, thanks! Otherwise, I think I can get to it later this week.

closed 2017-10-17 by todbot