todbot/blink1 #219: Need new Python version on PyPi

jimfmunro, 2015-03-19:
I'm confused as it looks like this has been updated more recently than the "official" version fork by @salimfadhley . It looks like his fork has solved the missing __init__.py file for the python module.

But it has not been updated on PyPi since last June.  

Can someone step up and address this and get the latest version bumped so it can be imported as a module?  

Relates to: https://github.com/todbot/blink1/issues/140

todbot, 2015-03-19:
salim's version is the official one, this is just a mirror.  As far as I know there's nothing new in this repo.

His version is only for Python 3 I think though.

salimfadhley, 2015-03-23:
I've not done any work on this in months - is there a specific change you need? I can have a go.

franc0is, 2015-09-05:
@salimfadhley see https://github.com/todbot/blink1/issues/140. The library no longer works on python2 because of it. After pip install, import still fails:
```
+ ~/Work/francois-poneys/blink1 on master* $ python
Python 2.7.10 (default, Jun 10 2015, 19:42:47) 
[GCC 4.2.1 Compatible Apple LLVM 6.1.0 (clang-602.0.53)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
+>>> import blink1
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named blink1
```

salimfadhley, 2015-09-05:
I have not played with this for ages - I can give it a once over sometime
this week.

jjshoe, 2016-04-19:
getting bit by this as well :(

todbot, 2016-04-19:
hey @salimfadhley, I could make these changes if you point me at the basics of how to generate content for PyPi (or make me a contributor if PyPi supports such a concept, my PyPi id is 'todbot')

jjshoe, 2016-04-19:
An additional complication is he doesn't appear to have issues turned on for that repo?

salimfadhley, 2016-04-19:
Okay - I can make Todd the project owner and also do a release. It won't be
until late tonight.

Todd what is your username

todbot, 2016-04-19:
Hi @salimfadhley, my PyPi username is "todbot".

salimfadhley, 2016-04-19:
Todd, you have been added as a project publisher.

I've just joined a team within Bank of America that's a keen user of Blink1
mk2 lights. I'm going to add my two to the collection and hopefully use
this as an excuse to improve our software.

Now remind me - whose version is the official version these days? My
preference would be to kill off my version of the software since I think
it's confusing to have somebody other than the blink1 company holding the
official version of the software.

Have you any idea what it is I'm supposed to be fixing. Sorry - totally out
of it! :-)

todbot, 2016-04-20:
Hi Salim,
No problem, I understand about lack of time!  
I think there are two main issues: 
1. The version on PyPi is from your fork of this repo, but issues with that version come here, and 
2. This library is Python 3 only and many people want it to work with Python 2.

My questions to you are:
1. Where is there a good doc on how to publish to PyPi?
1. How hard is it to make this library work with Python 2?

Thanks!

salimfadhley, 2016-04-20:
I will make a script with all the publishing logic

Making if python compatible should be easy as long as the USB libraries ate
functionally the same.

jklmnn, 2017-07-05:
Is this still active?
There are several bugs in the python3 blink1 module from pypi that are fixed in this repo. Will there be an update in pypi?

salimfadhley, 2017-07-05:
Sorry - I've not really touched it in ages. I've been meaning to!

I suggest that you take a look at the source code. Perhaps you'd like to
have a go at it!

jklmnn, 2017-07-07:
Well the bugs I mean are all fixed in this repo. All that would be required is to update your repo with the contents of this one or use this one to create the pypi package (which I would prefer since this repo is also on the [Blink1 library page](https://blink1.thingm.com/libraries/)).