todbot/blink1 #220: Blink1Control not working properly on Windows 7+

jcap360, 2015-03-26:
I have been using the controller for months now and I never had a problem, even with the December 4th update to Blink1Control.  But since about a month ago I have not been able to even open up the controller main window.  Every time I do it freezes the window on a white screen and does not respond.  This was not a problem for me as it was still running my IFTTT recipes just fine as long as I kept the controller running on my computer in the background and didn't try to open the controller main window.  But now it stopped working with my recipes completely.  I even tried using the old controller and that worked fine but it did not respond to IFTTT recipes.  Someone able to help?

asoria73, 2015-08-06:
This exactly is happening to me since I upgrade to Windows 10

todbot, 2017-10-17:
This should be fixed in Blink1Control2 : https://github.com/todbot/Blink1Control2/releases

closed 2017-10-17 by todbot

jjs68, 2017-12-04:
I had same issue (white screen in app) on 2010 Pro, to resolve, did a right click on Blink1Control2.exe and pick troubleshoot compatibility, then pick recommended setting, click test the program, click the 'more' then run anyway, then next and save

todbot, 2017-12-04:
Hi @jjs68,
Thank you for that work-around suggestion. It could be the clue as to it's happening for some people.   I've opened on issue for Blink1Control2 as issue https://github.com/todbot/Blink1Control2/issues/68

When you say "2010 Pro" do you mean Windows 7?

jjs68, 2017-12-06:
hi @todbot I was running fine on windows 7, was when moved to windows 10 that had to use compatibility mode