todbot/blink1 #227: Feature request: blink1-tool to show version of itself

taa1, 2015-05-02:
Hi,

I'm running blink1-tool on a Raspberry Pi, though this request is probably appropriate for all platforms.

`blink1-tool --version` currently shows the version of the firmware. Can you have it also show the version of the blink1-tool itself?

todbot, 2015-05-02:
`blink1-tool --version`  does show the version of blink1-tool.  How did you get blink1-tool and what are you seeing?

taa1, 2015-05-12:
You're right, `--version` does show its own version, when invoked under newer versions of blink1-tool. From [#228](https://github.com/todbot/blink1/issues/228) I was running an old version. Sorry to have wasted your time. I was getting `firmware version: 203` and not the blink1-tool version.

closed 2015-05-12 by taa1