todbot/blink1 #228: blink(1) cycles through colors on powerup but then ignores commands from blink1-tool

taa1, 2015-05-02:
* Raspberry Pi
* blink1-tool v1.97
* firmware on blink(1) is 203
* `51-blink1.rules` was installed to /etc/udev/rules.d/ and works successfully (I don't have to use sudo)

I'm able to send commands to the blink(1) but they seem to get ignored because the blink(1) doesn't react.

I've tried:
~~~~
pi@BlueDonut:~$ blink1-tool --on --id all
set dev:0 to rgb:0xff,0xff,0xff over 300 msec

pi@BlueDonut:~$ blink1-tool --random 10 --id all
random 10 times: 
0: 0/1 : be,a8,47 
1: 0/1 : 1a,f4,5b 
2: 0/1 : e2,86,40 
3: 0/1 : 4e,61,e7 
4: 0/1 : 52,72,1c 
5: 0/1 : 6b,ce,da 
6: 0/1 : 63,fa,37 
7: 0/1 : b1,0d,8d 
8: 0/1 : 35,55,5e 
9: 0/1 : c9,39,20 
pi@BlueDonut:~$ 
~~~~
But no lights come on.

What am I doing wrong? It's been a while since I playe-, I mean, WORKED, with this blink(1). Is there such a thing as resetting its configuration to factory defaults? I don't remember the last thing I doing with it.

todbot, 2015-05-02:
There aren't any factory defaults you need to worry about with blink(1).  It's too simple for that. 

A few notes:
- In general you should specify the `--id` option before any command like `--random`
- The command format for `--random` requires an equal sign: `--random 10`. Not sure how you're getting it to parse it as 10, unless you're running a very old version
- Just to verify, what does `blink1-tool --id all --on` do?

todbot, 2015-05-02:
Also, what is the output for the following commands:
```
$ which blink1-tool
$ blink1-tool --version
$ ldd blink1-tool  [using the full path of the which above]
```

taa1, 2015-05-12:
Hi Tod,

Thanks for your prompt reply, and I would have responded earlier but for some reason github didn't send me notifications of your response.

I'm embarrassed. I thought I had the latest version on my RPi but it's an old one:

````
-rwxr-xr-x 1 root root 1020394 Nov 28  2012 blink1-tool
````
I've installed the latest from source and it's working fine now. But to answer your questions, in case someone else does the same (stupid) thing I did of running an old version. All of the following were done on the old version:

>Just to verify, what does blink1-tool --id all --on do?

Still no light, but verbose says it's doing something:

````
blink1-tool --id all --on -v
deviceId[0] = 0
cached path = '20003349'
0: serial: '20003349'
set dev:0 to rgb:0xff,0xff,0xff over 300 msec
````

>$ which blink1-tool

I get: `/usr/local/bin/blink1-tool`

>$ blink1-tool --version

I get: `firmware version: 203`

>$ ldd /usr/local/bin/blink1-tool

I get: `not a dynamic executable`

closed 2015-05-12 by taa1

todbot, 2015-05-12:
Glad you figured it out, it was starting to stump me!