todbot/blink1 #229: Curiosity question about how blink1-tool retrieves firmware version

taa1, 2015-05-02:
Hi,

Running on Raspberry Pi.

Before I plugged in my blink(1), running `blink1-tool --version` said it couldn't find any devices. Once I plugged it in, `blink1-tool --version` worked OK and reported firmware 203. What I'm curious about is why I didn't have to specify *which* device ID to query. How does this work when there are multiple blink(1)'s attached that are potentially running different firmware?

todbot, 2015-05-02:
`blink1-tool --version` and `blink1-tool --fwversion` currently returns the firmware version of the "first" (lowest serial number) device plugged in.  I would consider it a bug that `--fwversion` doesn't obey the `--id` option, and I'll add that as a separate issue.

With no devices, `blink1-tool --version` will print out the version of the blink1-tool binary.  How are you obtaining the blink1-tool binary?

For example, here's one way to get and run blink1-tool:
```
# git clone https://github.com/todbot/blink1.git
# cd blink1/commandline
# make
[... make output elided ...]
# ./blink1-tool --version
blink1-tool version: v1.97-linux-x86_64
```

taa1, 2015-05-12:
From issue [#228](https://github.com/todbot/blink1/issues/228), I was running an old (circa 2012!) version of blink1-tool. I don't even remember where I got it from.

closed 2015-05-12 by taa1