todbot/blink1 #231: Need command-line REST server

todbot, 2015-05-08:
The REST server part of Blink1Control should be available as a separate headless, command-line program that is cross-platform and runs well in Linux and has minimal dependencies.  

There is a rudimentary one in blink1/commandline/server  but it is a bit stale and hard to add to.

leonboot, 2016-05-12:
I'd love to see a REST server with the same capabilities as the one in the Blink1Control GUI provides! We're using a Blink(1) for our development team and controlling it now requires one of our workstations. We've got a Raspberry Pi set up for the job, but like @todbot mentioned, it's rather stale and can only set the leds to an RGB value, not run any patterns like the REST server included in Blink1Control.

todbot, 2016-05-12:
Hi @leonboot,
Thanks for your comment.  
What are some features and requirements you'd like to see in such a server?
What other features would you like?  
Is NodeJs too heavy-weight for your applications?

For instance, here are two NodeJs-based servers:
- https://www.npmjs.com/package/blink1-status-hub
- https://www.npmjs.com/package/node-blink1-server

leonboot, 2016-05-12:
Hi @todbot!

I had found both the servers you mentioned, and basically what I'm missing in both these and the blink1-tiny-server from this repo are the pattern-related commands. This would make for a much more diverse set of signals we can display! For instance, having it turn red when our CI build fails is completely doable with either of the servers, but finding an easily distinguishable set of colors only goes so far; having our Slack bot play the 'policecar' pattern when someone posts "There's free cake in the lunch room!" is much more recognizable but will only work if the Blink(1) is hooked up to one of our workstations running Blink1Control 😜

todbot, 2016-05-12:
haha, awesome. okay, I'll put patterns at the top of the list!

leonboot, 2016-05-12:
@todbot Awesome!