todbot/blink1 #232: Blink(1) freezes, can't click anything, this happens most of the time.

futurebird, 2015-05-10:
I have an iMac from 2008 with the latest OSX from apple. My Blink is one of the early models, not gen 2.

I can control the light from the command line without issue. The app is useless. It boots and then is frozen right away.

Once or twice I was able to make changes but it froze soon after.

I want to use IFTTT.... are there any alternaives?

todbot, 2015-05-11:
Hi,
If you have an original blink(1) with metal top and you're having issues with Blink1Control, there are two things you can do. First is try using the previous iteration of Blink1Control that was made for the original blink(1): http://thingm.com/blink1/downloads/Blink1Control-0.997-applescript.zip

The other is you can contact us via at blink1 @ thingm.com and we can do more debugging.  The original blink(1) was more sensitive to what kind of USB hubs it was plugged into, so we'll likely just end up shipping you a replacement blink(1)mk2.

futurebird, 2015-05-13:
Using the old software is working OK for me... though my long term goal is to put it on the USB port of a router running ddwrt.

closed 2015-08-13 by todbot