todbot/blink1 #237: Blink(1) Control Font Rendering Issue

Varachack, 2015-06-23:
Hello.

I'm running Windows Vista Home Premium, and upon launching the Blink(1) control application every time, some of the characters within the text in the program are improperly rendered and displayed as small squares; distorting the words.

When I click on one of the main tabs that is displaying the rendering bug within the text, the text on the selected tab is corrected, and upon deselection, the issue returns to the text.

Below is a high-res image displaying the issues.

[image]

todbot, 2015-10-01:
Hi,

We’ve recently released an updated version of Blink1Control, v1.98.
Hopefully it will address the issues you’ve been experiencing.

You can download the new version here:
http://blink1.thingm.com/blink1control/
Or on this repo's releases page

dunk8888, 2016-02-03:
hi ime having a blank screen issue,ive installed latest blink 1 control but it starts white screen,ive not been able to use it yet,whats the fix,do i need update java or somthing,ime windows 7 32bit

todbot, 2017-10-17:
Fixed in Blink1Control2 : https://github.com/todbot/Blink1Control2/releases

closed 2017-10-17 by todbot