todbot/blink1 #238: no blink(1) devices found on Linux mint

JulienCharge, 2015-06-26:
Hello,
I just installed a freshly the latest version of linux mint LDME.

I have to install the library
sudo apt-get install libusb-1.0-0-dev

I recovered the source git repository
git clone https://github.com/todbot/blink1.git

I always ARPES compilation this error
sudo ./blink1-tool --green
No. blink (1) devices found

sudo ./blink1-tool --list
No. blink (1) devices found

Do you have an idea of the problem?

Thank you for your answer.

Julian

todbot, 2015-06-26:
Hi Julian,
What does the output of "lsusb" say on your system?

JulienCharge, 2015-06-26:
julien@rainf01 ~ $ sudo lsusb 
Bus 003 Device 002: ID 8087:8001 Intel Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 5986:024b Acer, Inc 
Bus 001 Device 003: ID 8087:07dc Intel Corp. 
Bus 001 Device 002: ID 046d:c050 Logitech, Inc. RX 250 Optical Mouse
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

JulienCharge, 2015-06-26:
thank you,

For your answer,
As above, the result of lsusb.

Is it possible to blink1 not compatible with my pc?

Is this the driver can be a problem?

Do you have devised for my problem?

thank you

good day

todbot, 2017-10-17:
Hi @JulienCharge 
Apologies for the long delay getting back to you.  Did you ever find out the issue you were seeing?  I forgot to ask initially: Are you using a "blink(1)" or a "blink(1) mk2" device?