todbot/blink1 #239: Blink(1) doesn't work...

aiden56, 2015-06-26:
So I just got a blink(1) and I just downloaded the software...

1. IFTTT can't verify my IFTTT key, so I can't connect the channels.
2. If I click one of the patterns and select play (even if it's a static colour) the blink(1) just flashes erratically random colours.
3. If I turn on auto start-up on login and open minimised, the application doesn't seem to work at all...

aiden56, 2015-06-26:
Also, it doesn't detect a serial number for the blink(1) if that helps? Is it faulty? Should I return it to Amazon for an exchange?

todbot, 2015-06-26:
Hi Aiden,
Yes, your blink(1) sounds like it might be faulty.  I would try a different USB port or USB hub just to eliminate the current port, cable or hub you're using.  If it still doesn't show up, you should do an exchange with Amazon.

closed 2015-08-13 by todbot