todbot/blink1 #240: no blink(1) devices found, but device is present

wsv-accidis, 2015-07-03:
I am trying to use the blink(1) as a status light for an embedded Linux device, a Synology NAS (DS415+). I have managed to build and install libusb-1.0.9 and the blink(1) shows up just fine in lsusb:

    Bus 001 Device 005: ID 27b8:01ed
    Device Descriptor:
      bLength                18
      bDescriptorType         1
      bcdUSB               2.00
      bDeviceClass            0 (Defined at Interface level)
      bDeviceSubClass         0
      bDeviceProtocol         0
      bMaxPacketSize0         8
      idVendor           0x27b8
      idProduct          0x01ed
      bcdDevice            0.02
      iManufacturer           1 ThingM
      iProduct                2 blink(1) mk2
      iSerial                 3 20004DA9
      bNumConfigurations      1
      Configuration Descriptor:
        bLength                 9
        bDescriptorType         2
        wTotalLength           41
        bNumInterfaces          1
        bConfigurationValue     1
        iConfiguration          0
        bmAttributes         0x80
          (Bus Powered)
        MaxPower              120mA
        Interface Descriptor:
          bLength                 9
          bDescriptorType         4
          bInterfaceNumber        0
          bAlternateSetting       0
          bNumEndpoints           1
          bInterfaceClass         3 Human Interface Device
          bInterfaceSubClass      0 No Subclass
          bInterfaceProtocol      0 None
          iInterface              0
            HID Device Descriptor:
              bLength                 9
              bDescriptorType        33
              bcdHID               1.01
              bCountryCode            0 Not supported
              bNumDescriptors         1
              bDescriptorType        34 Report
              wDescriptorLength      24
             Report Descriptors:
               ** UNAVAILABLE **
          Endpoint Descriptor:
            bLength                 7
            bDescriptorType         5
            bEndpointAddress     0x81  EP 1 IN
            bmAttributes            3
              Transfer Type            Interrupt
              Synch Type               None
              Usage Type               Data
            wMaxPacketSize     0x0008  1x 8 bytes
            bInterval               1
    Device Status:     0x0001
      Self Powered

And in /dev/bus/usb/devices:

    T:  Bus=01 Lev=02 Prnt=02 Port=02 Cnt=02 Dev#=  5 Spd=12   MxCh= 0
    D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 8 #Cfgs=  1
    P:  Vendor=27b8 ProdID=01ed Rev= 0.02
    S:  Manufacturer=ThingM
    S:  Product=blink(1) mk2
    S:  SerialNumber=20004DA9
    C:* #Ifs= 1 Cfg#= 1 Atr=80 MxPwr=120mA
    I:* If#= 0 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbhid
    E:  Ad=81(I) Atr=03(Int.) MxPS=   8 Ivl=1ms

Clearly the device is present but completely unrecognized by blink1-tool. I tried using the blink1raw tool but the hidraw* devices are not present on the device.

Any suggestions on how to troubleshoot this?

todbot, 2015-07-03:
Hi!
On Linux, you'll need to either run the programs as the "root" user (with "sudo" for instance), or install a udev-rule like the one in this directory: https://github.com/todbot/blink1/tree/master/linux

wsv-accidis, 2015-07-03:
Thanks for your quick reply. I am logged in as root already (I know this is bad practice, but it makes things a lot easier when tinkering). Maybe I still need to add a rule.

Is udev something I need to have installed in order to use blink1? I can't find a clear reference on whether the Linux on my NAS has it by default. Any easy way to check?

todbot, 2015-07-03:
Did you recompile the blink1-tool binary too?

wsv-accidis, 2015-07-04:
I haven't been able to compile blink1-tool on the Synology, I downloaded a binary release for Linux-i686. If I could build it, I could try the HIDDATA/HIDRAW modes. Right now I think it's complaining about missing dependencies.

Cross-compiling would also be an option I guess but I've no clue how that works as of yet... as I mentioned, bit of a newbie, and I'm thinking the issues I am having compiling blink1-tool on the NAS are probably mostly due to that.

FWIW, here are the compiler errors I'm getting:

    Building for OS=linux BLINK1_VERSION=v0-linux-x86_64 USBLIB_TYPE=HIDAPI
    cc -DUSE_HIDAPI -I./hidapi/hidapi  `pkg-config libusb-1.0 --cflags` -fPIC -std=gnu99  -g -DBLINK1_VERSION=\"""v0"-linux-"x86_64""\" -c blink1-tool.c -o blink1-tool.o
    cc -DUSE_HIDAPI -I./hidapi/hidapi  `pkg-config libusb-1.0 --cflags` -fPIC -std=gnu99  -g -DBLINK1_VERSION=\"""v0"-linux-"x86_64""\"  -g ./hidapi/libusb/hid.o blink1-lib.o  `pkg-config libusb-1.0 --libs` -lrt -lpthread -ldl blink1-tool.o -o blink1-tool
    ./hidapi/libusb/hid.o: In function `read_thread':
    /root/blink1/commandline/hidapi/libusb/hid.c:760: undefined reference to `libusb_handle_events_completed'
    /opt/lib/gcc/i686-linux-gnu/4.2.1/../../../../i686-linux-gnu/lib/librt.so: undefined reference to `__librt_enable_asynccancel@GLIBC_PRIVATE'
    /opt/lib/gcc/i686-linux-gnu/4.2.1/../../../../i686-linux-gnu/lib/libpthread.so: undefined reference to `__libc_sigaction@GLIBC_PRIVATE'
    /opt/lib/gcc/i686-linux-gnu/4.2.1/../../../../i686-linux-gnu/lib/librt.so: undefined reference to `__librt_multiple_threads@GLIBC_PRIVATE'
    /opt/lib/gcc/i686-linux-gnu/4.2.1/../../../../i686-linux-gnu/lib/librt.so: undefined reference to `__librt_disable_asynccancel@GLIBC_PRIVATE'
    /opt/lib/gcc/i686-linux-gnu/4.2.1/../../../../i686-linux-gnu/lib/libpthread.so: undefined reference to `_h_errno@GLIBC_2.0'
    /opt/lib/gcc/i686-linux-gnu/4.2.1/../../../../i686-linux-gnu/lib/libpthread.so: undefined reference to `__on_exit@GLIBC_PRIVATE'
    /opt/lib/gcc/i686-linux-gnu/4.2.1/../../../../i686-linux-gnu/lib/libdl.so: undefined reference to `_dl_open@GLIBC_PRIVATE'
    /opt/lib/gcc/i686-linux-gnu/4.2.1/../../../../i686-linux-gnu/lib/libdl.so: undefined reference to `_dl_close@GLIBC_PRIVATE'
    /opt/lib/gcc/i686-linux-gnu/4.2.1/../../../../i686-linux-gnu/lib/libpthread.so: undefined reference to `_errno@GLIBC_2.0'
    collect2: ld returned 1 exit status
    make: *** [blink1-tool] Error 1

Note, the Synology doesn't have cc, it's actually using gcc. Not sure if this matters.

todbot, 2015-07-05:
Ahh, yes, blink1-tool has dependencies that are difficult to satisfy on the smaller embedded Linuxes.

There is a stripped-down tool called "blink1-mini-tool"  that uses libusb-0.1 in `blink1/commandline/blink1-mini-tool`.  Change into that directory and do a `make`.  That has many fewer dependencies and so should compile if your libusb also has the older function names.

Are you compiling these by sshing into the Synology and compiling directly on it?  If so, that's cool, I need to get me one of these.

wsv-accidis, 2015-07-07:
Blink1-mini-tool compiled and runs just fine, thank you so much for your help! Now I just need to write the software to update the light from what's going on in the NAS. I'm probably going to start off using it as a status light for CrashPlan.

Yeah, the embedded Linux is surprisingly competent once you install a development toolchain and package manager. I can wholeheartedly recommend it. Mine is the DS415+.

closed 2015-07-27 by wsv-accidis