todbot/blink1 #241: BlinkControl gives device ID but says no blink(1) attached in Windows 7

kevintynfron, 2015-07-20:
I've got my blink plugged in with  Blink(1) Control running and I can't get it to do anything.

BlinkControl says "no Blink(1) found), but if I go to settings, I can set it to use the specific device id.
I've got 2 blinks and the device id changes according to which one I've got plugged in. This suggests that it's seeing the blink in one way, but not in another part of the program.

I've tried the blinks in other machines and they work ok, pointing to my machine being the culprit.

I'm running Windows 7 with patches and service packs up to date on a Lenovo T520. 

[image]

todbot, 2015-07-20:
Hi,
This is very weird.  The only time I've seen this happen is when two apps are trying to control the same blink(1) device.  Could you be running two copies of Blink1Control?  Or do you have a script or daemon running that controls blink(1)?

kevintynfron, 2015-07-21:
No other blink software running, but by terminating processes that seemed to have anything to do with a usb port one by one, I have found the problem.
It was some software called "Upsilon 2000" which is to monitor a ups supply from [Zigor](http://www.zigor.com/uk/index.php?option=com_content&id=201&Itemid=210&lang=en)
There's a task called `usbmate.exe` for communicating with the ups. If that task is stopped, BlinkControl works fine.

closed 2015-07-21 by kevintynfron

todbot, 2015-07-21:
Oh interesting, `usbmate.exe` was grabbing any non-keyboard HID device (instead of grabbing by VID/PID).  

Thanks for the update, in case it comes up again.