todbot/blink1 #242: OS X 10.11 DP 4 and frozen Blink1Control app

lo5co, 2015-07-22:
Just updated to 10.11 DP 4 and the whole interface of Blink1Control app (except the settings icon) is frozen.

todbot, 2015-10-01:
Hi,

We’ve recently released an updated version of Blink1Control, v1.98.
Hopefully it will address the issues you’ve been experiencing.

You can download the new version here:
http://blink1.thingm.com/blink1control/
Or on this repo's releases page

darrenofkent, 2015-11-01:
I have 1.98 installed and the GUI is frozen overtime I launch blink(1)control ever since I upgraded my OS to 10.11 El Capitan.

todbot, 2015-11-02:
Hi Darren,
Can you download the debug version at https://github.com/todbot/blink1/releases/tag/v1.98,  unzip the bundle, open Terminal and run the app by typing
 `./Blink1Control-debug.app/Contents/MacOS/Blink1Control`
This will print out a bunch of debug message and hopefully give an idea as to what's going on.  Thanks!

darrenofkent, 2015-11-02:
Here it is:

QML debugging is enabled. Only use this in a safe environment.

running osFixes

settingsLoad:  "/Users/darrenkent/.config/ThingM/Blink1Control.ini"

"log:1446503284896: settingsLoad: settings loaded"

"log:1446503284896: settingsLoad: iftttKey='fd93b6e420002E1D'"

blink1IndexStr:  "0"

blink1Index:  "0"

"log:1446503284901: input name:Name1 iftttTime: \u009E nowSecs: 1446503284"

"log:1446503284964: refreshBlink1State: --- starting. count:1"

"log:1446503284970: refreshBlink1State: open & cached blink1 #0"

"log:1446503284970: refreshBlink1State: opening for use blink1Index:0"

"log:1446503284970: refreshBlink1State: connected, blink1serial='20002E1D'"

"log:1446503284970: refreshBlink1State: --- done. elapsedMillis:6"

qml: blink1 serial: 0 ,

"log:1446503287999: updateBlink1: fadeSpeed=400, #000000"

applicationStateChanged:  Qt::ApplicationState(ApplicationActive)

qt.network.ssl: QSslSocket: cannot resolve SSL_set_psk_client_callback

qt.network.ssl: QSslSocket: cannot resolve TLSv1_1_client_method

qt.network.ssl: QSslSocket: cannot resolve TLSv1_2_client_method

qt.network.ssl: QSslSocket: cannot resolve TLSv1_1_server_method

qt.network.ssl: QSslSocket: cannot resolve TLSv1_2_server_method

qt.network.ssl: QSslSocket: cannot resolve SSL_select_next_proto

qt.network.ssl: QSslSocket: cannot resolve SSL_CTX_set_next_proto_select_cb

qt.network.ssl: QSslSocket: cannot resolve SSL_get0_next_proto_negotiated

evname: "IFTTT snow tomorrow" evdate: 1446397854 recent: -1 last: 1446503284

evname: "IFTTT snow tomorrow" evdate: 1444066080 recent: -1 last: 1446503284

evname: "IFTTT snow tomorrow" evdate: 1444065965 recent: -1 last: 1446503284

evname: "Channel Activation" evdate: 1444064918 recent: -1 last: 1446503284

evname: "Channel Activation" evdate: 1444064918 recent: -1 last: 1446503284

evname: "IFTTT snow tomorrow" evdate: 1446397854 recent: -1 last: 1446503284

evname: "IFTTT snow tomorrow" evdate: 1444066080 recent: -1 last: 1446503284

evname: "IFTTT snow tomorrow" evdate: 1444065965 recent: -1 last: 1446503284

evname: "Channel Activation" evdate: 1444064918 recent: -1 last: 1446503284

evname: "Channel Activation" evdate: 1444064918 recent: -1 last: 1446503284

applicationStateChanged:  Qt::ApplicationState(ApplicationInactive)

evname: "IFTTT snow tomorrow" evdate: 1446397854 recent: -1 last: 1446503284

evname: "IFTTT snow tomorrow" evdate: 1444066080 recent: -1 last: 1446503284

evname: "IFTTT snow tomorrow" evdate: 1444065965 recent: -1 last: 1446503284

evname: "Channel Activation" evdate: 1444064918 recent: -1 last: 1446503284

evname: "Channel Activation" evdate: 1444064918 recent: -1 last: 1446503284

evname: "IFTTT snow tomorrow" evdate: 1446397854 recent: -1 last: 1446503284

evname: "IFTTT snow tomorrow" evdate: 1444066080 recent: -1 last: 1446503284

evname: "IFTTT snow tomorrow" evdate: 1444065965 recent: -1 last: 1446503284

evname: "Channel Activation" evdate: 1444064918 recent: -1 last: 1446503284

evname: "Channel Activation" evdate: 1444064918 recent: -1 last: 1446503284

evname: "IFTTT snow tomorrow" evdate: 1446397854 recent: -1 last: 1446503284

evname: "IFTTT snow tomorrow" evdate: 1444066080 recent: -1 last: 1446503284

evname: "IFTTT snow tomorrow" evdate: 1444065965 recent: -1 last: 1446503284

evname: "Channel Activation" evdate: 1444064918 recent: -1 last: 1446503284

evname: "Channel Activation" evdate: 1444064918 recent: -1 last: 1446503284

evname: "IFTTT snow tomorrow" evdate: 1446397854 recent: -1 last: 1446503284

evname: "IFTTT snow tomorrow" evdate: 1444066080 recent: -1 last: 1446503284

evname: "IFTTT snow tomorrow" evdate: 1444065965 recent: -1 last: 1446503284

evname: "Channel Activation" evdate: 1444064918 recent: -1 last: 1446503284

evname: "Channel Activation" evdate: 1444064918 recent: -1 last: 1446503284

darrenofkent, 2015-11-23:
So.... what do I do now?  Blink1Control app still won't work on El Capitan.

todbot, 2015-11-23:
Hi Darren,

The debug version shows the same behavior as the normal version?  Could you add a screenshot as to what you're seeing?  We've got a few Mac OS X 10.11 boxes here and can't replicate the problem.

darrenofkent, 2015-11-23:
When I open the app the GUI is frozen so a screenshot would not show that.
I have attached a short video of what happens when I try to make any
settings/changes with the GUI.  The blink1 seems to be triggering as it
should from my IFTTT channel but I can no longer make any changes or
settings adjustments through the GUI, as it is non-responsive.​
 IMG_0934.MOV
<https://drive.google.com/file/d/0BwHmi9QuDG6Pb3FIS1VSTC1takE/view?usp=drive_web>
​

todbot, 2017-10-17:
Fixed in Blink1Control2 : https://github.com/todbot/Blink1Control2/releases

closed 2017-10-17 by todbot