todbot/blink1 #244: Crashes when mac awakens from sleep (OSX 10.8)

pretty, 2015-08-05:
Whenever I open my computer or awaken it, the control app stalls and the wheel of death displays when I hover over the menu icon. The app requires the process to be killed and restarted every time.

I'm running OSX 10.8.5

-Stefan

todbot, 2015-08-06:
Hi, I'm looking into this now.

pretty, 2015-08-13:
Any luck? I've stopped using the application in the mean time.

todbot, 2015-10-01:
Hi,

We’ve recently released an updated version of Blink1Control, v1.98.
Hopefully it will address the issues you’ve been experiencing.

You can download the new version here:
http://blink1.thingm.com/blink1control/
Or on this repo's releases page

joschaschultze, 2015-10-06:
It is still crashing alle the time. I've set up a cronjob on my mac to check if it's still running but that's a really bad solution.

todbot, 2015-10-06:
That's frustrating.  Do you get any crash stack trace window popping up?  Or do you see a stack trace in syslog or Activity Monitor?

joschaschultze, 2015-10-06:
Here is an example. Crash at 00:09:21 and cronjob-start at 00:10:01

Oct  6 00:09:21 joschasltzesmbp com.apple.launchd.peruser.501[179] (com.thingm.Blink1Control.72192[83965]): Job appears to have crashed: Segmentation fault: 11
Oct  6 00:09:22 joschasltzesmbp.fritz.box ReportCrash[84242]: Saved crash report for Blink1Control[83965] version 1.98 (0) to /Users/joscha/Library/Logs/DiagnosticReports/Blink1Control_2015-10-06-000922_Joscha-Schultzes-MacBook-Pro.crash
Oct  6 00:09:22 joschasltzesmbp.fritz.box ReportCrash[84242]: Removing excessive log: file:///Users/joscha/Library/Logs/DiagnosticReports/Blink1Control_2015-10-05-175313_Joscha-Schultzes-MacBook-Pro.crash
Oct  6 00:10:01 joschasltzesmbp.fritz.box launchservicesd[96]: Application App:"Blink1Control" asn:0x0-30f30f pid:84254 refs=6 @ 0x7fa093d4a150 tried to be brought forward, but isn't in fPermittedFrontApps ( ( "LSApplication:0x0-0x1001 pid=85 "loginwindow"")), so denying. : LASSession.cp #1481 SetFrontApplication() q=LSSession 100004/0x186a4 queue
Oct  6 00:10:01 joschasltzesmbp.fritz.box WindowServer[116]: [cps/setfront] Failed setting the front application to Blink1Control, psn 0x0-0x30f30f, securitySessionID=0x186a4, err=-13066
Oct  6 00:10:01 joschasltzesmbp.fritz.box launchservicesd[96]: Application App:"Blink1Control" asn:0x0-30f30f pid:84254 refs=7 @ 0x7fa093d4a150 tried to be brought forward, but isn't in fPermittedFrontApps ( ( "LSApplication:0x0-0x1001 pid=85 "loginwindow"")), so denying. : LASSession.cp #1481 SetFrontApplication() q=LSSession 100004/0x186a4 queue
Oct  6 00:10:01 joschasltzesmbp.fritz.box WindowServer[116]: [cps/setfront] Failed setting the front application to Blink1Control, psn 0x0-0x30f30f, securitySessionID=0x186a4, err=-13066
Oct  6 00:10:01 joschasltzesmbp.fritz.box WindowServer[116]: CGXSetWindowListSystemAlpha: Invalid window 5425 (index 0/1)
Oct  6 00:10:01 joschasltzesmbp.fritz.box WindowServer[116]: CGXSetWindowListSystemAlpha: Invalid window 5426 (index 0/1)

todbot, 2017-10-17:
Should be fixed in Blink1Control2 : https://github.com/todbot/Blink1Control2/releases

closed 2017-10-17 by todbot