todbot/blink1 #245: Missing .lib file for the QT project (MSVC)

thorlist, 2015-08-18:
The QT .pro file says:

# for blink1-lib
# note: be sure to compile blink1-lib beforhard with:
# cd blink1/commandline && make lib"
# for Windows, that will build both MinGW & MSVC compatible libs

but when I make the commandline I don't get the .lib file needed for the QT build. I do get the .def, the .exe and the .dll, but have been unable to generate the .lib file from this.

Can you suggest how I would go about creating or downloading the .lib file that I need?

Thank you for your help!
Thor

todbot, 2015-08-18:
Hi,
Very strange.  What environment are you using to compile?  I use the shell from MinGW with the compiler from Visual Studio 12.

But if your goal is to just grab the Windows blink1-lib, perhaps the easiest would be to download the Blink1Control app zip and pull it from there.  It's sitting in the root directory of the zip.

todbot, 2015-08-18:
Oops, sorry, the .lib is not in the distribution, just the .dll.

What is the output you get do `cd commandline && make clean && make lib`?

todbot, 2015-08-18:
More specifically, do you have the `LIB.EXE` from Visual Studio in your path?  This is what is used to create the .lib & .def files.  With Visual Studio 12 installed, this is what I get:
```
$ which lib
/c/Program Files (x86)/Microsoft Visual Studio 12.0/VC/bin/lib.exe
```

thorlist, 2015-08-18:
Yep, that was the problem, lib.exe was not in the path.
Thank you!

closed 2015-08-18 by thorlist