todbot/blink1 #246: Blink(1) Control application GUI does not load

brent875, 2015-09-05:
Tried loading and unloading the application, and have not been able to successfully launch the blink(1) control application successfully.  Program attempts to launch, but does not render screen and stops.  Currently running Windows 8.1

todbot, 2015-09-05:
Hi,
Can you take a screenshot of what you're seeing? 
Also, are you running as a regular user or administrator?

brent875, 2015-09-05:
[image]

I've tried both regular user and administrator.

brent875, 2015-09-05:
[image]

I was running as an administrator to load the application.

todbot, 2015-10-01:
Hi,

We’ve recently released an updated version of Blink1Control, v1.98.
Hopefully it will address the issues you’ve been experiencing.

You can download the new version here:
 http://blink1.thingm.com/blink1control/
Or on this repo's releases page

labomat, 2015-10-08:
I have the same problem. BlinkControl starts and I can set modes (away etc) via Tray icon but GUi is just a blanc window.

I am using v1.98 on Windows 7 64bit.

Enursha, 2015-10-15:
Same issue here, is the a debug mode or logging we can enable?

todbot, 2015-10-16:
I've uploaded debug versions of v1.98 to the releases page: https://github.com/todbot/blink1/releases/tag/v1.98

There are 'debug' versions of the app for Mac & Windows.
On Windows, opening the app will open a cmd window with debug messages
On Mac, to see the debug messages, unzip the bundle, open Terminal in the same directory as the unzipped app, and run the app by typing `./Blink1Control-debug.app/Contents/MacOS/Blink1Control`

There's also a log you can turn on in Advanced Preferences (the gear icon), but I suspect this error is happening before the log is opened.

Please copy-n-paste the output of Terminal if you have this problem, I'm very curious.  I've not been able to replicate it on any of the computers I have access to.

labomat, 2015-10-16:
It is not working when installed in folder Program Files (x86) - works fine when opened from a "normal" folder (like "User/.../Downloads"). Debug see screenshot below: left side: not working (blank window).

[image]

Enursha, 2015-10-16:
The debug version is working fine. I tried uninstalling and reinstall the normal version and now it's also working without issue. This was an upgrade for me, not sure what version I previously had installed, guess a file or too weren't updating correctly.

Apologies, should of tried that to start with.

todbot, 2015-10-17:
Thank you @labomat, that's helpful. I think I might see where the problem is.
I'll get a second debug version to try out by Weds.
Quick question: you said this was Win7 x64 running as a normal user, what is your UAC setting?  (To find this, open Control Panel, search for "UAC", and click on "Change User Account Control setting". You'll be shown a slider with four positions ranging from "Always notify" to "Never notify".  Which position is your slider in?)  Thanks.

labomat, 2015-11-07:
Sorry for not writing back. Had to reinstall my PC due to a harddisc crash... Downloaded the Installer again and it works now. I run the standard UAC setting (second to top).

screentwelve, 2015-11-15:
I have the same problem. Just the frame of the window shows, and I can use the System Tray controls, but the program never loads. I am running Windows 7 Pro 64-bit. This is a fresh install on a brand new computer. I originally tried the Blink1Control 1.98 with installer and that didn't work, so I searched and found this forum. After uninstalling and reinstalling Blink1Control (tested again: didn't work) and uninstalling it again, I tried the zip file and running from Downloads folder and that didn't work. Then I tried the debug version from zip. Same issue, but this time it throws an error, even when running as Administrator. The dialog title is Microsoft Visual C++ Runtime Library. It says
Debug Error!
Program:
...Blink1Control-debug-win\Blink1Control-debug\Blink1Control.exe
R6010
- abort() has been called
(Press Retry to debug the application)

Attached is a screenshot of the command window debug info. The last line "qml: blink1 serial: 0 ," shows up after a few seconds, the same time as the error dialog, also attached.
[image]
[image]

todbot, 2017-10-17:
Fixed in Blink1Control2 : https://github.com/todbot/Blink1Control2/releases

closed 2017-10-17 by todbot

cameradoppia, 2021-01-21:
Hi there, sorry if I re-open this issue but I'm not able to solve the same bug.

I'm on iMac os HIghSierra (10.13.6)

Thanks in advance,
Daniele

todbot, 2021-01-21:
Hi @cameradoppia,  

Does the "Blink1Control2" menu exist and in that can you select the "Open Dev Tools" option?  That will open a "console" window that hopefully will have an error.

cameradoppia, 2021-01-28:
Hi, forgive my late reply: the Dev Tools works smoothly and I've solved by deleting a MAIL source that was giving an error, despite right credentials. Since I've done that, our lovely Blink1 is super fine. Thanks, best regards! Dan

todbot, 2021-01-28:
Thanks for the update!  Glad it's working now. Let me know if you have any further questions