todbot/blink1 #249: blink1-tool and blink(1) control do not play nice

baob, 2015-09-12:
Problem Scenario:  Use Blink(1) control to turn on the blink, then use blink1-tool to turn it off.

Expected Result: blink(1) turns off

Actual Result: 
```
$ blink1-tool --off
cannot open blink(1), bad id or serial number
```

Version numbers:  

Mac OS X:  10.9.5
blink(1) control: I can't find out, but downloaded within the last week.
blink1-tool: v0-macosx-x86_64, fw version: -1  / v0-macosx-x86_64, fw version: 204

I have a workaround (unplug and replug the blink(1)) but any help resolving this would be appreciated :smile:

todbot, 2015-09-12:
Hi,
You can't use both the Blink1Control app and the `blink1-tool` commmand-line program concurrently, since they both assume they can "own" the device.  As you've noticed, trying to do so will only work sporadically.

Your choices are either to: a) not use Blink1Control and `blink1-tool` and the same time or b) use `blink1control-tool` instead of `blink1-tool`.  You can find `blink1control-tool` in the releases page: https://github.com/todbot/blink1/releases (scroll down or search for "blink1control-tool")

Note that in order for blink1control-tool to work, you need to enable the API server in Blink1Control.  You can find the switch for that in the tray- /menubar-icon menu.

baob, 2015-09-12:
Thank you. I'll try `blink1control-tool` .... though if it needs the api server I may finish up using that directly.

Appreciate the snappy response  :+1:

todbot, 2015-09-12:
No problem. :)  Here's a doc explaining API server commands if you've not already come across it: https://github.com/todbot/blink1/blob/master/docs/app-url-api-examples.md

closed 2015-09-12 by todbot