todbot/blink1 #262: Make tutorial on using blink(1) mk2 with WS2812/Neopixel LEDs

todbot, 2015-12-16:
blink(1) mk2's secret feature is that it can control up to 16 WS812 LEDs.  We promised a tutorial on how to do this, but haven't done it yet.  The only thing we've done is:
http://todbot.com/blog/2014/09/10/blink16-prototype-is-woody-and-awesome/

kahnibus, 2016-04-06:
So cool!

Got something similar working but all I can do with the "extra" neopixel is mimic the Blink1 device it is connected to.  When I use the blink1-tool --list command the neopixel isn't listed.   What am I missing?

todbot, 2016-04-06:
Hi @kahnibus, 
In blink1-tool there is a '--led' (or '-l') option to specify which LED to address.  This number can range from 0-18.  So you have:

- `--led=0` means all LEDs
- `--led=1` means the first LED (top LED of the blink(1)
- `--led=2` is the bottom blink(1) LED
- `--led=3` is the 1st WS2812/Neopixel LED connected to a blink(1)
- `--led=18` is the 16th WS2812/Neopixel LED connected to a blink(1)

kahnibus, 2016-04-06:
Boosh!  It works!

Thank you soooo very much for the quick response!!!

Dave

*************************************
David G. Kahn
Campus Technology Services
School of Business
SUNY Oswego
315.312.6383

Campus Technology Services will never ask you to email us sensitive
personal information such as
​ a​
 password.
​P​
lease contact us if you are unsure if an email is genuine. (help@oswego.edu)

dunk8888, 2016-04-09:
So where is all this info so we dont keep asking? is it publishe anywhere?
it may help to make some pdfs of all the tools etc blink1-tool and
linkM-tool,i think ive found 3 main pdfs,ive loads of info  that ive found
its all over the place and sheets of small tutorials when i get time ile
try put it all together and publish it some where.

dunk8888, 2016-06-24:
Hi,just want to know what you used to send commands to the blinkM mk2
experamentation boards etc does it work with the blinkM mk1 arduino loader?
via cmd for e.g