todbot/blink1 #270: Blink(1) Control Frozen in Mac 10.11.1

nickwm, 2016-01-10:
I'm able to install Blink(1) Control and launch its Settings interface where my blink(1) is shown as connected. However, clicking anywhere on the Settings interface has no result. Clicking the tabs (IFTT, Tools, etc.) does not change the interface, and clicking the other controls (Available, Busy, etc) does not illuminate the USB light.

Settings did work for me at one time, but now I'm unable to get it out of this frozen state.

I'm running Mac OS X 10.11.1 (15B42) on a MacBook Pro (15-inch, Late 2011).

nickwm, 2016-01-10:
A couple more observations on this issue ...

- I am able to illuminate the blink by choosing Set to Busy/Available/Away from the Blink icon that appears to the left of the clock in the upper right of the screen.

- I am able to launch Advanced Preferences by clicking the gear icon on the Settings interface

todbot, 2017-10-17:
Fixed in Blink1Control2 https://github.com/todbot/Blink1Control2/releases

closed 2017-10-17 by todbot