todbot/blink1 #272: External software triggering Blink with NO RULES in the Blink tool!!

twannieguitar, 2016-01-14:
Hi

I have a program running called Roost Desktop Notifier which is an app to control a Nest thermostat.
When that program is runing in the systray and the thermostat starts heating the blink turns red. Whan ambient it flashes white and when away it is solid green.
Thi issue is, I HAVE NO RULES AT ALL in my blink tool and this still happens...

Anyone any idea?

hkallweit, 2016-01-16:
Seems to be intentional, see here:
http://richardeverett.com/Nest

-----------------------------------------------------------------------------------------------------------
Hey, it's your lucky day! Roost Desktop Notifier also supports the excellent full-colour blink(1) indicator light from ThingM.

Simply plug a blink(1) into a spare USB port on your PC and it will glow orange or blue when your Nest Thermostat is heating or cooling, or a pleasing green if you've earned a leaf.

It will flash warning yellow or emergency red if any of your Nest Protects have detected a smoke or CO warning or emergency.

twannieguitar, 2016-01-17:
Thanks, I already had contact with Richard to get the option to disable. Thanks again!

closed 2016-02-17 by todbot