todbot/blink1 #275: Blink1Control: add Slack integration

todbot, 2016-02-17:
Looks like it might not be too hard to add Slack integration, since IFTTT Slack channel doesn't support Slack event triggers.

Slack API: https://api.slack.com
Supported events: https://api.slack.com/events

Perhaps Blink1Control configuration is:
1. User inputs Web API auth token generated here: https://api.slack.com/web
2. User picks which events they want and what color patterns to trigger

conorsch, 2017-01-28:
Consider https://github.com/dschep/ntfy as a simpler alternative. Once configured, `ntfy send "foo"` will work just fine from the CLI, so it should be trivial to use with blink as well.

toddkurtz, 2017-01-28:
This would be cool! Can someone provide good instructions or an example of
this set up? This would give my blink a purpose.

Jammizzle, 2017-05-25:
Just looking to see if anyone has made any advancements on this? This sounds really interesting and I'd like to implement this in our business

dunk8888, 2017-05-25:
https://blink1.thingm.com/

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

jscontrust, 2017-06-08:
+1
Would love to have that. All my colleagues have their sound disabled for a good reason, so they do not see when I`m wrote a message to them

jnth, 2017-06-09:
If you do not want to explose a server, you can use the Slack Real Time Messaging system to create an interaction between Slack and the Blink1 device.

[Here is an example](https://gist.github.com/jnth/74dd378865ab16a737eb9fba1126de84) in Python 3.6.
It's based on the [Full Stack Python](https://www.fullstackpython.com/blog/build-first-slack-bot-python.html) blog.

jscontrust, 2017-06-12:
Ok, thank you. How would I implement something like that on OSX in an easy way?

donjuanmon, 2019-08-29:
+1 , would love to see this

KF99, 2020-02-11:
+1 Love both!

Skarlso, 2020-11-20:
https://github.com/todbot/Blink1Control2/pull/139
Blinker controller now can start a server and listen on the RTM with a classic slack app.