todbot/blink1 #279: /blink1/pattern/stop does not stop all patterns

agemmell, 2016-03-06:
The API docs imply that supplying no args to /blink1/pattern/stop means it will stop any/all currently playing patterns.  It does not.  Is this a bug?

From a terminal:
```
$> curl 'http://localhost:8934/blink1/pattern/play?pname=longRunningPattern'
$> curl 'http://localhost:8934/blink1/pattern/stop'
{
    "pname": "",
    "status": "pattern not found"
}
```
How do I stop any pattern(s) playing via the API without needing to know their name(s) beforehand?

BTW sending "off" (http://localhost:8934/blink1/off) while a pattern is running just has the effect of the LED fading to black/off briefly and then the previously running pattern just carries on as if nothing happened.  I just wanted to preempt anyone suggesting I run that command instead.

todbot, 2016-03-07:
Hi,  Currently you must specify which pattern to stop, just as you specified which pattern to play.  I would consider this an omission in the API and you are right there should be a way of stopping all patterns.

And `/blink/off` should stop pattern playback but does not.  That is a bug and code regression.

I'll look into pushing out an small update with these API server changes soon.  Thanks!

agemmell, 2016-03-07:
Awesome, thanks @todbot. I actually forgot to give you my system details (bad bug reporting, sorry).  FWIW: I'm using OSX 10.10.5 with BlinkControl Version: v1.98.

mightybs, 2017-03-06:
I just got the blink1 and was reading the documentation.  Sorry to see that these API calls still do not work.  I might be able to help with some PRs if you could point me in the right direction for the APIs that need repair.

todbot, 2017-03-06:
Development on this version of Blin1kControl has stopped.  
Please check out the new Blink1Control2 at https://github.com/todbot/Blink1Control2
It supports the APIs you're speaking of.

closed 2017-03-06 by todbot

mightybs, 2017-03-07:
Thank you!  Much appreciated.

dunk8888, 2017-03-08:
Is there blink1control for 32x systems windows/

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>