todbot/blink1 #281: 1.98-win: Unable to trigger IFTTT rules

m-p-3, 2016-03-10:
I've had to reinstall my OS (Windows 7 Enterprise) and since I am unable to get the IFTTT rules to actually trigger.

I've ensure I used the same rule names that I used in my IFTTT recipes, I updated my IFTTT blink(1) channel to the new IFTTT key the app provided me (I know I could have simply changed the host ID back to the original one.. too late I guess :/)

Whenever I trigger any recipes using IFTTT Maker, I see the recipe has been triggered on IFTTT, but it never actually reaches blink1control.

I ran the debug version and I saw this:
`qt.network.ssl: QSslSocket: cannot resolve TLSv1_1_client_method`
`qt.network.ssl: QSslSocket: cannot resolve TLSv1_2_client_method`
`qt.network.ssl: QSslSocket: cannot resolve TLSv1_1_server_method`
`qt.network.ssl: QSslSocket: cannot resolve TLSv1_2_server_method`
`qt.network.ssl: QSslSocket: cannot resolve SSL_select_next_proto`
`qt.network.ssl: QSslSocket: cannot resolve SSL_CTX_set_next_proto_select_cb`
`qt.network.ssl: QSslSocket: cannot resolve SSL_get0_next_proto_negotiated`

I'm not sure if it has anything to do with my problem.

One thing that is different is that I had to install .NET Framework 4.6.1 in order to get Paint.NET to work on my end, and apparently this release removes all previous frameworks releases present.

Anyone has an idea what I could do to get it to work?

m-p-3, 2016-03-10:
Hmm, I actually figured out that [this commit](https://github.com/todbot/blink1/commit/91514c04a6447dd2edff701314e3b270e5d73c0b) might be related to my problem.

Anything I can do, other than trying to compile a windows version from source? ;)

todbot, 2016-03-10:
Hi @m-p-3, we had an issue with the datacenter that IFTTT uses to bridge events to Blink1Control.  It's fixed now and events should be flowing as before.  Apologies for the downtime.

That commit you see is for an upcoming upgrade of the IFTTT channel that IFTTT will be rolling out shortly.  When the change happens, old versions of Blink1Control will still work.

todbot, 2016-03-10:
Also, the QSslSocket warnings are just warnings from Qt that we can't get rid of until we switch away from Qt.

m-p-3, 2016-03-10:
Oh so it wasn't something I could have fixed at this poiint! I'll test it out whenever I can.
Thanks for your support!

closed 2016-03-13 by todbot

QuadDepo, 2017-10-07:
testing this is the browser, the feed one doesnt response anything.
the api, one does? my ifttt actions aren't firing on the GUI

todbot, 2017-10-08:
Hi @QuadDepo,  the v1 of Blink1Control is no longer supported.  Please try the newer Blink1Control2, which you can find at https://github.com/todbot/Blink1Control2/releases