todbot/blink1 #282: Present Status of Skype for Business via Blink1Control?

vejuhust, 2016-03-10:
Hi @todbot 

One of my colleagues is using a product named [**Blynclight**](http://www.embrava.com/products/blync-light?variant=328886579) (previously **Blync**), which is similar to blink(1) but focus on Skype for Business. 
Do you have any plan to enable blink(1) to present status of Skype for Business or Skype via Blink1Control app?

[image]

todbot, 2016-03-10:
Hi @vejuhust, 
Yes, we're currently working on a major fixup of Blink1Control, and one of the features we're planning on having is responding Skype events via their web api.

vejuhust, 2016-03-10:
Cool!:yum: 
Apart from Skype, Skype for Business (previously Lync) will also be supported.:telephone_receiver: 
Have I got that right?

renenulsch, 2016-03-12:
Hi @vejuhust ,
I've received my first blink(1) some days before and the Lync/SfB integration was my first task. :-)

see https://github.com/renenulsch/Lync-Blink-Bridge for an very early version

todbot, 2016-05-02:
Hi @renenulsch,

This is great! How would one use this?  (I've not used Skype for Business)

vejuhust, 2016-05-04:
@todbot It can be implemented with pure Web now. 
FYR: [Web SDK](https://ucwa.skype.com/documentation/what-is-lync-ucwa-api), [Example](https://www.matthewproctor.com/skype-web-sdk-presence-dashboard-revisited)
Btw, I use Skype for Business everyday, and I'm so willing to be a tester for this feature.

PriyathamKommula, 2016-06-30:
@renenulsch 

How can I use this https://github.com/renenulsch/Lync-Blink-Bridge with my blink(1)?
Do I have to make any changes in my blink(1) settings?

todbot, 2016-06-30:
Hi @PriyathamKommula, I've not used this myself, but what I think you would do is:
1. Download the lync-blink-bridge.zip from the project's releases page:  https://github.com/renenulsch/Lync-Blink-Bridge/releases
2. Unzip the zipfile
3. Run the "LyncBlinkBridge.exe" executable.  

Note you should make sure that Blink1Control or other blink(1)-controlling software is not running at the same time.