todbot/blink1 #286: Blink(1) Control crashes, OS X El Capitan

colinmo, 2016-03-31:
Since updating to the latest release (v1.98, September 2015), my Blink(1) Control is regularly crashing. Checking the diagnostic report:

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY
...
Error Code:      0x00000014
Trap Number:     14

I haven't been able to determine any consistency or reliability in when it crashes; it just happens randomly.

colinmo, 2016-03-31:
Just crashed again, with exciting new exceptions!

Exception Type:        EXC_BAD_ACCESS (SIGBUS)
Exception Codes:       KERN_PROTECTION_FAILURE at 0x00000001047da321
Exception Note:        EXC_CORPSE_NOTIFY
...
Error Code:      0x00000007
Trap Number:     14

todbot, 2016-03-31:
Hi @colinmo,
That's frustrating. Could you try out the debug version at the bottom of this page: https://github.com/todbot/blink1/releases/tag/v1.98
And follow the instructions on how to turn on logging & see the debug messages?

colinmo, 2016-03-31:
Running now, debug messages filling the screen, will update when it crashes next.

colinmo, 2016-04-01:
Nothing yet, normally it's crashed a tonne by now. Maybe the debug did it? Coincidence?

colinmo, 2016-04-13:
Update. The debug version doesn't crash. The non-debug version does crash. Bleh. Does the non-debug one have crash/ logs anywhere I could dig up and send?

colinmo, 2016-04-13:
That said, even though it isn't crashing, I am getting 
```
url:  "imaps://imap.gmail.com:993/INBOX/"
checkMail: last value =  "error: Unsupported protocol"
curl failed
error:  Unsupported protocol
```
showing up in the logs, if that's any assistance.

todbot, 2017-10-17:
I believe this is fixed in Blink1Control2 : https://github.com/todbot/Blink1Control2/releases

closed 2017-10-17 by todbot

colinmo, 2017-10-17:
Thanks, will update and let you know when I get results.

colinmo, 2017-10-25:
Confirming, this has been running for a few days with zero crashes, bug is fixed.