todbot/blink1 #287: Blink1Control crashes immediately on startup (win7 x64)

TiredOfSomething, 2016-04-08:
I installed the software with no errors, but it immediately crashes on startup. I don't even get the screen loaded before it crashes. I've tried uninstall/reinstall. Win7 x64 Intel i5 plenty of RAM, nothing else really remarkable. I've had older versions that worked in the past, but I haven't used the device in a long time.

todbot, 2016-04-11:
Hi,
Does it crash with or without blink(1) plugged in?  Does the older version still work?  You can get older versions by scrolling down on this page: https://github.com/todbot/blink1/releases  Perhaps the version you last used was this one: https://github.com/todbot/blink1/releases/tag/v1.95

TiredOfSomething, 2016-04-11:
It crashes with or without the device connected. I tried back to 1.9 and it
would run the program, but not recognize the device. It said there was no
driver.

todbot, 2016-04-11:
When you say "it says there's no driver", you mean Windows, correct?  If so, and this is an original blink(1) (not a blink(1) mk2).  Then try putting a USB hub in between the blink(1) and your computer.  Early blink(1)s had issues with certain PC motherboards.

todbot, 2017-10-17:
I think this is fixed in Blink1Control2 : https://github.com/todbot/Blink1Control2/releases

closed 2017-10-17 by todbot