todbot/blink1 #288: python demo1.py issues

jjshoe, 2016-04-19:
https://github.com/todbot/blink1/blob/master/python/pypi/blink1_demo/demo1.py

I couldn't get it to run until I commented out L21 and L22. Then I get:

'''
blink(1) found
fading to #ffffff
Traceback (most recent call last):
  File "test.py", line 25, in <module>
    blink1.fade_to_rgb(1000, 255, 255, 255)
  File "/Users/joel/ansible-blink1/blink1/blink1.py", line 156, in fade_to_rgb
    return self.fade_to_rgb_uncorrected(fade_milliseconds, r, g, b, led_number=0)
  File "/Users/joel/ansible-blink1/blink1/blink1.py", line 152, in fade_to_rgb_uncorrected
    return self.write(buf)
  File "/Users/joel/ansible-blink1/blink1/blink1.py", line 124, in write
    buf)
  File "/Library/Python/2.7/site-packages/usb/core.py", line 1023, in ctrl_transfer
    buff = _interop.as_array(data_or_wLength)
  File "/Library/Python/2.7/site-packages/usb/_interop.py", line 97, in as_array
    a.fromstring(data) # deprecated since 3.2
TypeError: must be string or read-only buffer, not list
'''

I understand the python lib is supported by someone else, but if you look at their fork, you can't even create a issue there.

It seems like you should probably pull python from marketing materials until someone gets the time to clean up all the python issues. :\

todbot, 2016-04-19:
Are you on Python2 or Python3?  The library in PyPi is only for Python3 I believe.

But I agree. I'm working with Salim to get the PyPi library transferred over.  For now with Python2 you may have more luck with one of these libraries: https://github.com/todbot/blink1/tree/master/python/alternative_libraries

jjshoe, 2016-04-19:
Version 2, which is the default for OSX and Ubuntu 14.04LTS at the very least. Having it only work with 3 and not be backwards compatible seems like a dis service?