todbot/blink1 #291: Run blink1-tiny-server at start up (ubuntu mate / R-pi 2)

benvickers, 2016-09-10:
I would like to run the blink1-tiny-server on ubuntu mate (Raspberry Pi) at start up but am having some trouble. 

Have tried; 
- setting up as .deskstop autostart file 
- adding to rc.local 

Both as /bin application and to the file path (user/home/blink1/commandline/blink1-tiny-server). 

For some reason it never works - which is quite prohibitive to what I am trying to do with the blink.

todbot, 2016-09-11:
Hi Ben,

What exact command did you use for your rc.local?
I think you should be able to do it with something like:
```
 nohup blink1-tiny-server -p 8888 > /dev/null 2>&1 &
```
This assumes 'blink1-tiny-server' is in the path, otherwise put in full path. You may not need 'nohup'.

benvickers, 2016-09-11:
Thanks for swift reply! I tried this and seems not to work either. 

Something odd seems to be up with doing this on Ubuntu Mate on a Pi.
I've actually added blink1-tiny-server to /bin as an executable (I thought this might make it easier to exec). 

There seems to be a variant of issues before I even get to the auto start. 
1) Running blink1-tiny-server from path without Sudo, starts the server but I get: 
```
{
"uri":  "/blink1/on",
"result":  "blink1 on
on: couldn't find blink1",
"version": "0.94"
}
```
2) Same issue arises when running it in ref to the /bin command.  

It's possible I missing something super basic, I am by no means an expert in any of this and just tinkering around!

todbot, 2016-09-11:
Hi Ben,
Oh I see, the problem you're having is not you can't get blink1-tiny-server to run but that it can't find any devices without sudo?  That's a device permissioning problem and can be fixed with a "udev" rule.  Here is a file you can add to your system to enable non-root users to access blink(1) devices (the file has comments on how to install it):
https://github.com/todbot/blink1/blob/master/linux/51-blink1.rules

benvickers, 2016-09-11:
Hi again! :) 
I was going to write actually to say that I realised that was what was causing the device missing issue - this resolve that problem fortunately! 
But unfortunately I still cannot get it to autostart - either with rc.local or as a .desktop file. 
Checked chmod stat for all files - have tried various ways and locations for the blink1-tiny-server file itself, still nothing seems to work. 

Quite mystified at this point! 

Are there any other examples or ways of autostarting the server (or are there other linux servers i might try?) - many thanks in advance!

todbot, 2016-09-11:
Hi Ben,
Try out this script:
https://github.com/todbot/blink1/blob/master/linux/blink1-tiny-server-init.sh
To use it, do this in a terminal on your Pi:
```
sudo cp blink1-tiny-server-init.sh /etc/init.d/blink1-tiny-server
sudo update-rc.d blink1-tiny-server defaults
```
Then you can check it, start it up and stop it like this:
```
sudo service blink1-tiny-server status
sudo service blink1-tiny-server start
sudo service blink1-tiny-server stop
```

And the `update-rc.d` command above sticks the script in the init script startup area so it'll start up on boot.  

Let me know how this works out.

benvickers, 2016-09-11:
**Amazing thank you so much** - it works seamlessly :+1: :)

todbot, 2016-09-11:
Oh excellent.  I'll close this issue then.  Cheers!

closed 2016-09-11 by todbot