todbot/blink1 #292: Blink1 control crashes shows clear window in windows 7 32x

dunk8888, 2016-09-26:
blink1 control will not start,tryed older versions,wont work with or without blink1 mk1 pluged in.

todbot, 2016-09-26:
Hi,
Could you try out the new Blink1Control2 beta and see how that works for you?
More info about it here:
https://blog.thingm.com/2016/09/announcing-blink1control2-public-beta/

dunk8888, 2016-09-26:
crashes too

dunk8888, 2016-09-26:
debug version works now and again,

todbot, 2016-09-26:
That's very strange.  

What do you see when Blink1Control2 crashes?  
Are you using the installer or the zip file for Blink1Control2?

Is there anything about your Windows setup that is special? 
Are you running as an admin user or a regular user?

dunk8888, 2016-09-26:
Well a small box comes up with strange artifacts in it like its all muddled
up,then says wrong for my version of windows,like it needs 64x mine is
32x,also can you remember what dual diode is used on blink1 mk1 is it dual
cathode or dual anode?

todbot, 2016-09-26:
Oh you are running 32-bit Windows 7?  Okay, yes, both versions of Blink1Control will not work with that.  Apologies, I will make a build for that and get it out later this week.  

As for the LEDs in blink(1) mk2, they are not regular LEDs but rather "smart pixel" LEDs called WS2812B LEDs (aka "NeoPixel")  Do a search for either of those terms to see how they work.

dunk8888, 2016-09-26:
Hi well blink1 control used to work with 32x ime sure of it as it worked
before i reinstalled my windows 7.and i only run 32x,strange though.
with the blink 1 mk1 is the diode a dual cathode or dual anode ?

todbot, 2016-09-26:
I'll get a 32-bit version available later this week.

In the blink(1) mk1, the LED is common-cathode. (not dual because there's three elements, not two)   You can view the schematic here: https://github.com/todbot/blink1/blob/master/hardware/schematic/blink1_v1a_sch.pdf

dunk8888, 2016-09-26:
> schematic/blink1_v1a_sch.pdf

Hi i dont mean the led i mean the dual zener diode! the diode that conects
to the usb lines it says dual diode? but the "bom" says D1   3.6V dual
zener DIODE-2CA|ACSOT23 Dual DIODE " "Cathode-Anode|Anode-Cathode" "

--------------------------------------------
is it dual cathode or dual anode???

when i solder a dual anode the blink1 does nothing but when i solder a dual
cathode it *enumerates*
<https://www.google.co.uk/search?client=firefox-b&q=enumerates&spell=1&sa=X&ved=0ahUKEwiR3Ma9mK7PAhUJIMAKHej_AR8QvwUIGygA&biw=1440&bih=740>
on and off ? says unknown usb device,so its nearly working! but ime unsure
of whats wrong,ive made the brd from osh park and populated the board with
the correct componants but somthing is wrong? i had the same problem on a
*similar*
<https://www.google.co.uk/search?client=firefox-b&q=similar&spell=1&sa=X&ved=0ahUKEwjKuIbmma7PAhViB8AKHR4ZCrUQvwUIGygA>
circuite
<https://www.google.co.uk/search?q=circuite&client=firefox-b&biw=1440&bih=740&tbm=isch&tbo=u&source=univ&sa=X&ved=0ahUKEwjL8N_yma7PAhXKDcAKHQGzCtsQsAQILA>
and it turned out to be a missing cap but the caps are correct so ime
thinking its the dual 3.6 zener diode thats wrong,so i want to know if its
cathode or anode thats needed,when ive got the correct diode soldered then
i can debug the rest of the circuite and see what els could be wrong.
cheers

todbot, 2016-09-26:
Look at the schematic, it tells you how the zener diodes are configured and gives the exact part number.

closed 2016-09-26 by todbot

dunk8888, 2016-09-27:
Hi in the schematic it says the dual zener is 3.6v but the partnumber comes up in google as 2.7v and the datasheet says 2.7v. v-usb needs 3.6v diode doesnt it?

dunk8888, 2016-09-27:
Diodes Inc AZ23C2V7-7-F Dual Zener Diode, Common Anode, 2.7V 5% 300 mW SMT
3-Pin SOT-23

todbot, 2016-09-27:
you typed the part number in wrong

dunk8888, 2016-09-27:
Its the same as the schematic,it says on schematic AZ23C2V7-7-F   wich is
the 2.7v version. The 3.6V part number is AZ23C3V6-7-F  = I think you
probly used the 2.7v version in schematic as there wasnt a 3.6v to hand and
you wrote 3.6v next to it as not to get confused.Thats why i asked as the
part is labled AZ23C2V7-7-F but 3.6v is writen next to it.Oh i sorted the
blink1 control for my windows 7 32x it tuned out it was my graphics card
driver!,there was a log report and i saw somthing about 3d rendering error
or somthing like that,so i did an update on my graphics card driver and it
worked after that,but blink1 control2 doesnt as its 64 bit.
Ive made a blink1 clone on breadboard and works fine but the blink1 mk1 brd
from osh park doesnt work,ive tryed for days to sort it but it wont
(enumerate,install),it says uknown device and goes on off all the
time,maybe ive used the wrong type of dual zener diode everything looks
soldered properly, i had a problem with another v-usb circuit and was a cap
missing,this blink1 acts the same but ive installed the two caps one is
100n and 10uf,so ime sure its the dual zener thats the problem,its a comon
anode and its 3.6v so it should work,unless the other specs are wrong or
somthing,ime not sure of the mW,ime going to buy the ones you used as they
sell them on radio shack site,and hopefuly that will sort the
problem,cheers tod

todbot, 2016-09-27:
I'm not sure where you are seeing the 2.7V part number but it's a typo, it's a 3.6V zener.
In the schematic I linked previously:
https://github.com/todbot/blink1/blob/master/hardware/schematic/blink1_v1a_sch.pdf
The callout for D1 clearly says "3.6V dual zener D1: AZ23C3v6-7-F"
[image]