todbot/blink1 #296: blink1-tiny-server Syntax error: word unexpected (expecting ")")

moontear, 2017-01-22:
Trying to run blink1-tiny-server on a RaspberryPi 2 with Jessie.

Used the following to build:
```
pi@raspberrypi:/usr/local/blink1/commandline $ sudo make blink1-tiny-server
cc -DUSE_HIDAPI -I./hidapi/hidapi  `pkg-config libusb-1.0 --cflags` -fPIC -std=gnu99  -g -DBLINK1_VERSION=\"""v1.98"-linux-"armv7l""\" -I. -I./server/mongoose -c server/blink1-tiny-server.c
-o blink1-tiny-server.o
cc -DUSE_HIDAPI -I./hidapi/hidapi  `pkg-config libusb-1.0 --cflags` -fPIC -std=gnu99  -g -DBLINK1_VERSION=\"""v1.98"-linux-"armv7l""\" -I. -I./server/mongoose -c ./server/mongoose/mongoose.c
 -o ./server/mongoose/mongoose.o
cc -g ./hidapi/libusb/hid.o blink1-lib.o   ./server/mongoose/mongoose.o `pkg-config libusb-1.0 --libs` -lrt -lpthread -ldl -lpthread -ldl blink1-tiny-server.o -o blink1-tiny-server
```

When trying to run I get the following:
```
pi@raspberrypi:/usr/local/blink1/commandline $ sh blink1-tiny-server
blink1-tiny-server: 1: blink1-tiny-server: Syntax error: word unexpected (expecting ")")
```

Anything I'm doing wrong? Sudo also doesn't help, seems like a compilation issue to me?

todbot, 2017-01-22:
blink1-tiny-server (like blink1-tool) is an executable binary, not a shell script, so you just need to give its name without 'sh' to run it, like: `./blink1-tiny-server` if in the same directory or `/usr/local/blink1/commandline/blink1-tiny-server` when using an absolute path.

moontear, 2017-01-22:
Uhm... yes. I'll close this issue quickly so nobody sees ;-) Thanks!

I'll try to wrap the server into a real web server like nginx to get SSL support or maybe give the npm version a spin (https://www.npmjs.com/package/node-blink1-server).

closed 2017-01-22 by moontear

todbot, 2017-01-22:
oh yes that would be really cool!  The tiny-server is for people using blink(1) is very resource-constrained Linuxes like OpenWrt. I've not worked much on making a stand-alone webserver for it. Can't wait to see what you come up with and if I can help, let me know!