todbot/blink1 #297: Blink1_pyusb.py only finds first attached device

fullphat, 2017-01-28:


fullphat, 2017-01-29:
Update: I've fixed blink1.py and blink1_pyusb.py to accept an optional "unit" argument on creation.  At the moment this is just expected to be an **int** ranging from 0..n where 'n' is the total number of blink1 devices connected less one.  It's a bit clunky, but it does work - for example:

```
blink1 = Blink1()
other_blink1 = Blink1(1)

blink1.fade_to_rgb( 1000,  255,255,255 )
other_blink1.fade_to_rgb( 1000,  255,255,255 )
```

Happy to share code - I don't really understand this git stuff :-)

todbot, 2017-01-29:
Hi @fullphat,
Thanks!  We're trying to move the python library to its own repo.  It's not fully fleshed out yet, but could you add this pull request to https://github.com/todbot/blink1-python?

fullphat, 2017-01-29:
Will do - I'm very new to git so i'll read up on what needs doing with regards to pull requests! :)