todbot/blink1 #299: Blink1Control quit unexpectedly on launch. Mac OS X El Capitan.

snaketown, 2017-02-15:
Hi, 

I've been using my Blink1 Control software successfully for the last 3 or 4 months but last week it began quitting unexpectedly on launch. It now won't stay open for more than a second of two before quitting. I didn't install any new software or change anything for that matter.

I tried trashing the application and re-downloading but this didn't help.

I had a look at the crash report but I don't really understand what the issue is. Something to do with the email maybe? Thread 14? I've pasted the relevant parts of the crash report below.

The notification light is super helpful so any help getting it working again would be greatly appreciated.

Process:               Blink1Control [1085]
Path:                  /Applications/Blink1Control.app/Contents/MacOS/Blink1Control
Identifier:            com.thingm.Blink1Control
Version:               1.98 (0)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Blink1Control [1085]
User ID:               501

Date/Time:             2017-02-16 09:56:54.054 +1100
OS Version:            Mac OS X 10.11.3 (15D21)
Report Version:        11
Anonymous UUID:        9070DA02-E5FD-B921-2C07-68520E0D808F

Time Awake Since Boot: 3500 seconds

System Integrity Protection: enabled

Crashed Thread:        14  Email

Exception Type:        EXC_BAD_ACCESS (SIGBUS)
Exception Codes:       KERN_PROTECTION_FAILURE at 0x0000700000637f38
Exception Note:        EXC_CORPSE_NOTIFY

VM Regions Near 0x700000637f38:
    Stack                  00007000005b5000-0000700000637000 [  520K] rw-/rwx SM=COW  thread 13
--> STACK GUARD            0000700000637000-0000700000638000 [    4K] ---/rwx SM=NUL  stack guard for thread 14
    Stack                  0000700000638000-00007000006ba000 [  520K] rw-/rwx SM=COW  thread 14

Thread 14 Crashed:: Email
0   org.qt-project.QtCore         	0x0000000103319b07 0x103076000 + 2767623
1   org.qt-project.QtCore         	0x00000001033255c1 0x103076000 + 2815425
2   org.qt-project.QtCore         	0x000000010332258a 0x103076000 + 2803082
3   org.qt-project.QtCore         	0x00000001033222d6 0x103076000 + 2802390
4   org.qt-project.QtCore         	0x00000001033255c1 0x103076000 + 2815425
5   org.qt-project.QtCore         	0x000000010332258a 0x103076000 + 2803082
6   org.qt-project.QtCore         	0x00000001033222d6 0x103076000 + 2802390
...
THIS CONTINUES UNTIL LINE 511 BUT I HAVE OMITTED BECAUSE IT SEEMED IRRELEVANT.

Thread 14 crashed with X86 Thread State (64-bit):
  rax: 0x00007fc28d320b74  rbx: 0x0000000000000002  rcx: 0x0000000000000006  rdx: 0x00007fc2893bd218
  rdi: 0x00007fc2893bdad4  rsi: 0x00007fc28d320b74  rbp: 0x0000700000638100  rsp: 0x0000700000637ed0
   r8: 0x00007000006b7860   r9: 0x0000000000000000  r10: 0x00007fc28d320b6e  r11: 0x00007000006b7860
  r12: 0x0000000000000004  r13: 0x000000000000045a  r14: 0x0000000000000002  r15: 0x0000000000000005
  rip: 0x0000000103319b07  rfl: 0x0000000000010202  cr2: 0x0000700000637f38
  
Logical CPU:     2
Error Code:      0x00000006
Trap Number:     14

Binary Images:
       0x1022d5000 -        0x102473fff +com.thingm.Blink1Control (1.98 - 0) <96FBCEEA-B270-31C9-94E9-915DA553ACFD> /Applications/Blink1Control.app/Contents/MacOS/Blink1Control
       0x1024a6000 -        0x1024aaff7 +libBlink1.dylib (0) <A62AC519-02F9-3111-B245-66F997648372> /Applications/Blink1Control.app/Contents/Frameworks/libBlink1.dylib
       0x1024b8000 -        0x1029ccff7 +org.qt-project.QtWidgets (5.5 - 5.5.0) <DF74BFA6-B274-3FC1-889B-AE8CCAF166E0> /Applications/Blink1Control.app/Contents/Frameworks/QtWidgets.framework/Versions/5/QtWidgets
       0x102b06000 -        0x102f99ff7 +org.qt-project.QtGui (5.5 - 5.5.0) <B7C3E3B9-4466-38CD-8939-D4728537A4C0> /Applications/Blink1Control.app/Contents/Frameworks/QtGui.framework/Versions/5/QtGui
       0x103076000 -        0x103584fff +org.qt-project.QtCore (5.5 - 5.5.0) <E1A2B357-93A4-395A-AB30-D5D8E85031E1> /Applications/Blink1Control.app/Contents/Frameworks/QtCore.framework/Versions/5/QtCore
       0x103616000 -        0x1038fdff7 +org.qt-project.QtQuick (5.5 - 5.5.0) <713CC4AC-430F-31E9-9DF6-BC7A2186996A> /Applications/Blink1Control.app/Contents/Frameworks/QtQuick.framework/Versions/5/QtQuick
       0x1039a9000 -        0x103d06ff7 +org.qt-project.QtQml (5.5 - 5.5.0) <99737A2B-07BD-3468-8467-11553ABD3A1E> /Applications/Blink1Control.app/Contents/Frameworks/QtQml.framework/Versions/5/QtQml
       0x103d94000 -        0x103eb0fff +org.qt-project.QtNetwork (5.5 - 5.5.0) <F675D0EA-1DBF-3C19-9399-EEA0FBB8592C> /Applications/Blink1Control.app/Contents/Frameworks/QtNetwork.framework/Versions/5/QtNetwork
       0x103eea000 -        0x103eeefff  com.apple.agl (3.3.1 - AGL-3.3.1) <8B380B9E-3E00-3065-842E-41F4421D8D5A> /System/Library/Frameworks/AGL.framework/Versions/A/AGL
       0x10419a000 -        0x1041cffff +org.qt-project.QtPrintSupport (5.5 - 5.5.0) <5116A322-CA11-3C44-9029-81FB53B51F3F> /Applications/Blink1Control.app/Contents/Frameworks/QtPrintSupport.framework/Versions/5/QtPrintSupport
       0x106311000 -        0x1063b8fff +libqcocoa.dylib (0) <FB09B9DA-5F7C-377B-972A-8C591D1DD085> /Applications/Blink1Control.app/Contents/PlugIns/platforms/libqcocoa.dylib
       0x1063f1000 -        0x106458fff +org.qt-project.QtDBus (5.5 - 5.5.0) <4B97B481-FD03-3312-B33F-54A9E902DC85> /Applications/Blink1Control.app/Contents/Frameworks/QtDBus.framework/Versions/5/QtDBus
       0x1093b9000 -        0x1093c4fff  com.apple.iokit.IOHIDLib (2.0.0 - 2.0.0) <E71FA0E7-EAC9-3810-B519-DDDE32692B72> /System/Library/Extensions/IOHIDFamily.kext/Contents/PlugIns/IOHIDLib.plugin/Contents/MacOS/IOHIDLib
       0x1093d5000 -        0x1093dfff7 +libqdds.dylib (0) <CDF80BEA-3CFA-3F81-8B9C-931CB12FEC75> /Applications/Blink1Control.app/Contents/PlugIns/imageformats/libqdds.dylib
       0x1093e4000 -        0x1093ebff7 +libqgif.dylib (0) <51F4DF97-9999-3FA3-9C6D-E0CDDCC0D47F> /Applications/Blink1Control.app/Contents/PlugIns/imageformats/libqgif.dylib
       0x1093ef000 -        0x1093f7fff +libqicns.dylib (0) <5CBF3967-9FAA-3F70-9691-1E82F5E38315> /Applications/Blink1Control.app/Contents/PlugIns/imageformats/libqicns.dylib
       0x1093fc000 -        0x109402ff7 +libqico.dylib (0) <FDD111B5-638F-3B84-A39C-CD2A3A41D837> /Applications/Blink1Control.app/Contents/PlugIns/imageformats/libqico.dylib
       0x109407000 -        0x10948efff +libqjp2.dylib (0) <A8404AA8-77B5-3DEF-920A-52D34989402B> /Applications/Blink1Control.app/Contents/PlugIns/imageformats/libqjp2.dylib
       0x10949c000 -        0x1094dafff +libqjpeg.dylib (0) <F4EB5413-63BB-3529-AC0D-E98C3328303C> /Applications/Blink1Control.app/Contents/PlugIns/imageformats/libqjpeg.dylib
       0x1094e0000 -        0x109530ff7 +libqmng.dylib (0) <D2A1A517-7C7D-3BAC-A600-709E023B7D5B> /Applications/Blink1Control.app/Contents/PlugIns/imageformats/libqmng.dylib
       0x109538000 -        0x10953cff7 +libqtga.dylib (0) <3D814602-B539-38C7-BD2F-AF056BF92907> /Applications/Blink1Control.app/Contents/PlugIns/imageformats/libqtga.dylib
       0x109540000 -        0x1095a2ff7 +libqtiff.dylib (0) <23A39D56-6B66-3624-ADE2-93685C91FDE5> /Applications/Blink1Control.app/Contents/PlugIns/imageformats/libqtiff.dylib
       0x1095aa000 -        0x1095aeff7 +libqwbmp.dylib (0) <ADCD45B7-E6E9-34CC-94BA-E5974E938703> /Applications/Blink1Control.app/Contents/PlugIns/imageformats/libqwbmp.dylib
       0x1095b2000 -        0x109611fff +libqwebp.dylib (0) <5ABD01B1-35ED-3B4B-92DF-624BC710475A> /Applications/Blink1Control.app/Contents/PlugIns/imageformats/libqwebp.dylib
       0x1096f5000 -        0x1096f9ff7 +libqtquick2plugin.dylib (0) <AEA027FB-4571-3440-BCAC-B2AADF4A4F6F> /Applications/Blink1Control.app/Contents/PlugIns/quick/libqtquick2plugin.dylib
       0x1096fc000 -        0x1097bbff7 +libqtquickcontrolsplugin.dylib (0) <0CD0F14C-0903-3CFC-A250-04A4836B94B9> /Applications/Blink1Control.app/Contents/PlugIns/quick/libqtquickcontrolsplugin.dylib
       0x1097d2000 -        0x1097e3fff +libqquicklayoutsplugin.dylib (0) <A93668B7-A64E-31B1-8726-3408315E8BAE> /Applications/Blink1Control.app/Contents/PlugIns/quick/libqquicklayoutsplugin.dylib
       0x1097eb000 -        0x1097efff7 +libwindowplugin.dylib (0) <6B736192-04D5-3967-B281-52053B7537B9> /Applications/Blink1Control.app/Contents/PlugIns/quick/libwindowplugin.dylib
[... cut ...]
       0x109921000 -        0x10992bfff +libqmlfolderlistmodelplugin.dylib (0) <5A247C85-0510-36C8-B171-8FBFC6DFA189> /Applications/Blink1Control.app/Contents/PlugIns/quick/libqmlfolderlistmodelplugin.dylib
       0x109932000 -        0x109937fff +libqmlsettingsplugin.dylib (0) <96D7CC08-F183-30AA-BBB6-01837F492B0A> /Applications/Blink1Control.app/Contents/PlugIns/quick/libqmlsettingsplugin.dylib
       0x10cd9a000 -        0x10cdfcfff  com.apple.driver.AppleIntelHD4000GraphicsMTLDriver (10.12.13 - 10.1.2)

todbot, 2017-02-16:
Hi,
That's very strange, and yes I agree with your assessment that  it looks like it's something to do with the email system.  You may be able to fix it by deleting all your email entries and recreating them.  Perhaps there's something in the config that got corrupted and that could fix it.

But it was mysterious errors like this that caused me to recreate Blink1Control using a different app framework that is being updated more and has a greater number of libraries. The new app is called Blink1Control2 and you can download it here:
https://github.com/todbot/Blink1Control2/releases

Unfortunately the config file format is different so you'll have to re-enter your details.

snaketown, 2017-02-16:
Thanks Todbot, you mention that:

> You may be able to fix it by deleting all your email entries and recreating them

Will that be possible if I can't open the app? Is there a plist file or similar I can edit outside the application? If so where would I find this file?

Thanks

todbot, 2017-02-16:
Oh yes, of course, apologies.  For Blink1Control, the config is stored in 
```
~/.config/ThingM/Blink1Control.ini
```
If you're familiar with the Terminal, you can just `rm` that file.  In the Finder, use the "Go" -> "Go to Folder..." menu and enter "~/.config/ThingM" into the dialog box.  You can then trash the file like normal.

snaketown, 2017-02-16:
Thanks. I trashed the file and the application runs again and I can manually turn the light on and off. But... When I recreate the Blink1Control.ini file by re-configuring the email, the program instantly quits again. Its like the existence of that file causes the application to crash... Any thoughts?

snaketown, 2017-02-17:
Are there any other files the software creates similar to the Blink1Control.ini file when installing for the first time?

I would like to go through and delete everything related to the application then try re-installing.

todbot, 2017-10-17:
I believe this is fixed in Blink1Control2 : https://github.com/todbot/Blink1Control2/releases

closed 2017-10-17 by todbot