todbot/blink1 #301: blink1MK2 works when directly plugged in, but disconnect after a few seconds when using a USB cable

dswhite42, 2017-03-19:
My blink1MK2 works fine when directly plugged into a USB port.  I can run test patterns through Blink(1) Control, make API calls, run blink1control-tool, etc.  However when I plug the device into a USB port via a short (3ft) USB extension cable, the device is only recognized for a short time (between 5-60 seconds).  The test patterns will run in Blink(1) control, but the device itself will suddenly stop responding (usually freezing in whatever color it was last displaying.  After a few seconds Blink(1) control will change from "blink(1) connected" to "no blink(1) found"

I have tried this:
- with multiple Windows 10 PCs
- using multiple USB cables of different lengths
- and trying multiple USB ports per PC

The disconnection after 5-60 seconds over a USB cable is consistent across all tests.

This is Blink(1) Control 1.98 on Windows 10.

todbot, 2017-03-19:
Hi @dswhite42,
I have seen a few cases where a USB cable is just no good for data transfer, but since you you've tried multiple cables, it seems it's not that.  Contact us via email at blink1 at thingm dot com and we'll see about arranging a replacement blink(1) to be sent to you.

dswhite42, 2017-04-06:
Hello, Tod.  I ended up just getting a replacement blink(1) from where I originally bought it, and the replacement appears to be working properly.  Thanks.

closed 2017-04-06 by dswhite42