todbot/blink1 #302: Blink(1) crashes when checking Gmail

atr911, 2017-03-22:
It has been working for months now and all of a sudden it crashes every time it checks gmail.  If I change the login details so that it doesn't have the correct password anymore, it runs (but obviously doesn't check the mail).  I've tried uninstalling and reinstalling and I've tried using the beta (which doesn't seem to check email at all). 

Thoughts?  It's creating an issue at my work as we rely on this light to work properly.

Thanks!

todbot, 2017-03-22:
Hi, Please try out the new Blink1Control2 application to see if it works for you.   Its repo is here: https://github.com/todbot/Blink1Control2 and you can download it here: https://github.com/todbot/Blink1Control2/releases

The layout is a bit different and the config isn't compatible, but it is the codebase we're doing updates on now.

atr911, 2017-03-23:
I've downloaded it and it doesn't crash but I can't seem to get it to check
the email account properly. Suggestion?

dswhite42, 2017-04-06:
Possibly related to #300 ?

dswhite42, 2017-04-06:
I've had to stop using the GMail checker completely.  I wrote a Python script to do the mail checking instead, which uses the blink1control-tool.exe to trigger the blink(1).

atr911, 2017-04-06:
How do I do that? lol

dswhite42, 2017-04-06:
Well, here's how I did it:

https://gist.github.com/dswhite42/b9f3a94dcfeb8fc8c06c8dba4aad59da

todbot, 2017-10-17:
We also improved the email checking functionality a bit as of 2.0.2. 
https://github.com/todbot/Blink1Control2/releases

closed 2017-10-17 by todbot