todbot/blink1 #310: blink1-tool  and serial number

atsju, 2017-06-13:
Option -d or --id won't work with serial when the serial number is not one of a known mk2.
I use my own serial (as it is open hardware I made my own boards) so I cannot use blink1-tool. I will have to use serial from your boards ...

todbot, 2017-06-13:
Please give more details. There is nothing in blink1-tool that restricts which serial numbers to use.  The only restrictions are the serial number must be in the USB device descriptor and it must be a 4-byte hex string (e.g. "ABCD1234" is valid but "JKLMZXCV" is not valid)

Also, do you have a link to your board firmware and design files?  Perhaps there's an issue with your firmware I can help with.

atsju, 2017-06-13:
OK so here is my HW design : 
[circuitMaker](https://workspace.circuitmaker.com/Projects/Details/Julien-STAUB/blink1-pannel)
I only changed the LED because I did not found the other but it has the same driver.

The firmware is your 
[hex](https://github.com/todbot/blink1/blob/master/hardware/firmware_mk2/blink1mk2.production.20140901.hex) not even recompiled

I used serial [reference](https://github.com/todbot/blink1/blob/master/hardware/firmware_mk2/blink1mk2-serialization-2017Q1.txt)

If I use the serial 0x01000000 (I noticed you used big endian it may not be best choise because Microchip MPLAB X IPE can be used to generate serials but in little endian. But I respect and this choice and have no problem with it).
So with this serial the --list won't show (mk2) AND -d 0x01000000 won't work.

I solved the problem using same serial as reserved for seeedstudio  Q1 2017 (I made my PCB there too ;) but I do not find this very clean. Note also I do not plan to sell this boards (at least not more than a few units) so this serial problem is not crytical.