todbot/blink1 #312: Zapier Integration?

jon64digital, 2017-08-08:
Hi,

Are you planning on offering a "no code" Zapier integration? I have just bought one of your USB lights and I'm trying to find a way to get our chat software (drift.com) to make the light flash when we have somebody online trying to talk to us. It seems like Drift is compatible with Zapier and Blink(1) is compatible with IFTTT, but not vice versa.

Also, Is it possible to get anything to communicate directly with the device or does the communication always have to be through the client running on the desktop?

jon64digital, 2017-08-24:
Is this thing even supported any more? Massively disappointed since I bought mine. Service is terrible, integrations are lacking. Won't be buying any more.

todbot, 2017-08-24:
Hi Jon,
Apologies, I've been ramping up development again after having to take a several month hiatus.
I've been looking into Zapier and I think we can add it in as an event source.

todbot, 2017-08-24:
btw, the app development is at https://github.com/todbot/Blink1Control2 .  This repo is for the C & Java tools & libraries now.

jon64digital, 2017-08-24:
Hi Tod,

I'm glad to hear that, because there's nothing like it on the market that I know of.

I bought one, with the simple aim of getting it to flash every time somebody started a chat on our website chat widget (by drift.com). I had hoped that there would be a simple way of calling a simple REST API or similar that would make it flash but it seems that it's rather more complicated than that so I resorted to  looking into the integrations to see if that would simplify it. The chat widget has a Zapier integration.

I think part of the problem is that there are so many SDKs, so many Github repositories that it's hard to know where to start and the unit itself doesn't come with any kind of manual or quick-start guide to point you in the right direction, I was left floundering, trawling through the various SDK docs to see if there was any way I could get it working, but I'm a web developer, not a C++ guy or anything like that so I hit my limits pretty quickly.

I realize that this probably isn't the place for feedback like this and I didn't realize I was posting it in the wrong place, but I really do hope we can get something working as I had intended to buy 3 or 4 more for our sales guys once we'd got it working.

todbot, 2017-08-24:
If you'd like a simple REST API directly to blink(1), you can try the `blink1-tiny-server` commandline program.  It's very simple, minimal footprint HTTP server that talks directly to the blink(1) hardware.  You can download a version of it for you OS from the releases page and read more about it here: https://github.com/todbot/blink1/tree/master/commandline/server

Alternatively, you can run the Blink1Contol2 GUI application and enable its more full-featured REST API server.  That sounds like overkill for what you want. But the docs for how it works are:
https://github.com/todbot/blink1/blob/master/docs/app-url-api-examples.md
https://github.com/todbot/blink1/blob/master/docs/app-url-api.md

And apologies for the state the docs and thanks for your patience.  Everything is in a bit of a scattered state.  Part of the development effort I'm working towards once I get the next update to Blink1Control2 out is to fully move out the GUI app stuff from this repo.  (I came from the cvs & svn world where you had one big repo, but git is more about lots of little repos. I wish I would've done that originally for blink1)

todbot, 2017-09-02:
Closing this issue here.  I just set up preliminary Zapier support for blink(1).  See todbot/Blink1Control2#47 for more info.

closed 2017-09-02 by todbot