todbot/blink1 #313: Daemon mode with support for IFTTT

mhalano, 2017-08-30:
The idea here is quite simple and very useful: has a daemon running which can receive commands from IFTTT like Blink1Control2 do. The idea is keep the Blink1 on my Raspberry Pi and blink the light if some online event occurs. I know there's a lot of ways to achieve this, like APIs, connecting remotely, but none methods support IFTTT.

todbot, 2017-08-30:
There is a very basic Python script https://github.com/todbot/blink1/blob/master/python/blink1-ifttt-simple.py that does some of what you're asking but it's not very userfriendly.

Are you running standard Raspbian on your Pi?

mhalano, 2017-08-30:
I'm running Ubuntu Server, so no GUI.

todbot, 2017-08-30:
Ubuntu Server on a Raspberry Pi?  Interesting. Didn't know they supported that.  The script referenced above should still work.  But I agree, a proper daemon for IFTTT->blink(1) is warranted.

mhalano, 2017-08-30:
There's no officially support. I needed a little voodoo. I want to use Ubuntu Snappy Core, but I would need to build Snaps for a lot of things.

mhalano, 2017-08-30:
May be we could handle this on blink1-tiny-server?

todbot, 2017-08-30:
blink1-tiny-server is meant to be a minimal-dependency server for micro-Linuxes (like routers and such) but still also work for modern Linux, Windows, macOS.  So it needs to be in C and small and cross-platform.  Just getting the mongoose http framework working everywhere for it is a pain, let alone a JSON parser, or OAuth2 client (which we'll need when IFTTT finally updates our API to them)

If we can assume a full Linux, then this IFTTT->blink(1) thing could be written in a much faster-to-write in language like Python or Node.

taa1, 2017-10-04:
Is Blink1Control GUI the only way to generate the IFTTT blink id needed for https://github.com/todbot/blink1/blob/master/python/blink1-ifttt-simple.py?

todbot, 2017-10-04:
Hi @taa1, it's the preferred way, but you can use the blink(1) serial number doubled (e.g. if the serial number is `2345ABCD` then you can use `2345ABCD2345ABCD`.

closed 2017-10-17 by todbot