todbot/blink1 #314: Serial Number now responds with #2

BrianAker, 2017-09-15:
Subject says it all. 

host: ~ $ blink1-tool  --version
blink1-tool version: v1.98-macosx-x86_64, fw version: -1
host: ~ $ blink1-tool  --fwversion 
cannot open blink(1), bad id or serial number

Any ideas?

todbot, 2017-09-15:
Hi,
Which blink(1) do you have the original or the mk2?
Does the blink(1) work otherwise?

BrianAker, 2017-09-17:
The description on the Amazon order:

"ThingM blink(1) USB RGB LED BLINK1MK2"

	— Brian

乌

todbot, 2017-09-17:
Hi,
Does the blink(1) work otherwise?
Also, which OS are you on?

BrianAker, 2017-09-18:
OSX  10.12.6 (16G29)

It responds to some events, but not others. If I go into the control panel and set it to “meeting” it might work for a little while.

It turns off after a little while though. The USB bus is not showing that it is detaching.

From System Information:

blink(1) mk2:

  Product ID:	0x01ed
  Vendor ID:	0x27b8
  Version:	0.02
  Serial Number:	200061F6
  Speed:	Up to 12 Mb/sec
  Manufacturer:	ThingM
  Location ID:	0x00223000 / 3
  Current Available (mA):	500
  Current Required (mA):	120
  Extra Operating Current (mA):	0

On other note, if I leave it one “use first” it will work for a while, but if I select “2”, it shuts down.

	— Brian

> 
> Hi,
> Does the blink(1) work otherwise?
> Also, which OS are you on?

乌

todbot, 2017-09-18:
Only one executable at a time can talk to a blink(1) device.  Thus, one cannot run `blink1-tool` while running the Blink1Control app.  

If you'd like the functionality of `blink1-tool` while still running the Blink1Control app, you can use `blink1control-tool`, a command-line program that works similarly to `blink1-tool` but actually makes REST API calls to the Blink1Control app.

Apologies if this wasn't clear in the description of the tools.

BrianAker, 2017-09-18:
I turned off Blink1Control app and tried the command line tool again:

blink1-tool --list
blink(1) list: 
id:0 - serialnum:2 

~ $ blink1-tool --fwversion -v
deviceId[0] = 0
cached list:
0: serial: '2' 'USB_27b8_01ed_223000'
openById: 0
id:0 - firmware:204 serialnum:2 

Still no luck, the serialnum does not seem to be reporting itself correctly. 

> Only one executable at a time can talk to a blink(1) device. Thus, one cannot run blink1-tool while running the Blink1Control app.
>
> If you'd like the functionality of blink1-tool while still running [...]

乌

dunk8888, 2017-09-19:
Are you using a 3.0 hub? try an old 2.0 hub.

BrianAker, 2017-09-26:
The hub is off the monitor and has been the same hub it has worked on for years (all other devices are the same). I did move it though just to be sure, same thing.

乌

BrianAker, 2017-09-27:
Well good news/bad news.

I bought a second blink and once I inserted it, checked its firmware/serial number.

Everything checked out fine.

So I inserted the older blink and it started providing a sane serial number again.

> Only one executable at a time can talk to a blink(1) device. Thus, one cannot run blink1-tool while running the Blink1Control app.
>
> If you'd like the functionality of blink1-tool while still running [...]

乌

todbot, 2017-10-17:
That is so weird.  Well let us know if you have any future issues.  If the original one turns out to have a fault, email us at blink1 at thingm.com and we can arrange a replacement.

closed 2017-10-17 by todbot