todbot/blink1 #315: USBError: [Errno 13] Access denied (insufficient permissions) on pypi with macOS

andkon, 2017-09-27:
Hey there. Whether in or out of a virtualenv, I get the following error when attempting to run any of the sample code:

```
---------------------------------------------------------------------------
USBError                                  Traceback (most recent call last)
<ipython-input-3-ab702b03cf46> in <module>()
      3 
      4 with blink1() as b1:
----> 5     b1.fade_to_color(100, 'navy')
      6     time.sleep(10)
      7 

~/Developer/.virtualenvs/blinkpy3/lib/python3.6/site-packages/blink1/blink1.py in fade_to_color(self, fade_milliseconds, color)
    160         """
    161         red, green, blue = webcolors.name_to_rgb(color)
--> 162         return self.fade_to_rgb(fade_milliseconds, red, green, blue)
    163 
    164     def off(self):

~/Developer/.virtualenvs/blinkpy3/lib/python3.6/site-packages/blink1/blink1.py in fade_to_rgb(self, fade_milliseconds, red, green, blue, led_number)
    150     def fade_to_rgb(self,fade_milliseconds, red, green, blue, led_number=0):
    151         r, g, b = self.cc(red, green, blue)
--> 152         return self.fade_to_rgb_uncorrected(fade_milliseconds, r, g, b, led_number=0)
    153 
    154     def fade_to_color(self, fade_milliseconds, color):

~/Developer/.virtualenvs/blinkpy3/lib/python3.6/site-packages/blink1/blink1.py in fade_to_rgb_uncorrected(self, fade_milliseconds, red, green, blue, led_number)
    146         tl = fade_time & 0x00ff
    147         buf = [REPORT_ID, action, red, green, blue, th, tl, led_number]
--> 148         return self.write(buf)
    149 
    150     def fade_to_rgb(self,fade_milliseconds, red, green, blue, led_number=0):

~/Developer/.virtualenvs/blinkpy3/lib/python3.6/site-packages/blink1/blink1.py in write(self, buf)
    118                                (3 << 8) | REPORT_ID,  # (3==HID feat.report)
    119                                0,
--> 120                                buf)
    121 
    122     def read(self):

~/Developer/.virtualenvs/blinkpy3/lib/python3.6/site-packages/usb/core.py in ctrl_transfer(self, bmRequestType, bRequest, wValue, wIndex, data_or_wLength, timeout)
   1032                 and rqtype != util.CTRL_TYPE_VENDOR:
   1033             interface_number = wIndex & 0xff
-> 1034             self._ctx.managed_claim_interface(self, interface_number)
   1035 
   1036         ret = self._ctx.backend.ctrl_transfer(

~/Developer/.virtualenvs/blinkpy3/lib/python3.6/site-packages/usb/core.py in wrapper(self, *args, **kwargs)
    100         try:
    101             self.lock.acquire()
--> 102             return f(self, *args, **kwargs)
    103         finally:
    104             self.lock.release()

~/Developer/.virtualenvs/blinkpy3/lib/python3.6/site-packages/usb/core.py in managed_claim_interface(self, device, intf)
    165 
    166         if i not in self._claimed_intf:
--> 167             self.backend.claim_interface(self.handle, i)
    168             self._claimed_intf.add(i)
    169 

~/Developer/.virtualenvs/blinkpy3/lib/python3.6/site-packages/usb/backend/libusb1.py in claim_interface(self, dev_handle, intf)
    809     @methodtrace(_logger)
    810     def claim_interface(self, dev_handle, intf):
--> 811         _check(self.lib.libusb_claim_interface(dev_handle.handle, intf))
    812 
    813     @methodtrace(_logger)

~/Developer/.virtualenvs/blinkpy3/lib/python3.6/site-packages/usb/backend/libusb1.py in _check(ret)
    593             raise NotImplementedError(_strerror(ret))
    594         else:
--> 595             raise USBError(_strerror(ret), ret, _libusb_errno[ret])
    596 
    597     return ret

USBError: [Errno 13] Access denied (insufficient permissions)
```

I'm on macOS 10.12.6, and am running Python 3.6.2.

todbot, 2017-10-17:
Hi @andkon,
Apologies for the delay in responding to this.  Did you ever find the cause of this issue you were seeing?  One thing I can think of: are you running the Blink1Control application at the same time? The Python library expects to have exclusive control of the blink(1) device. If you want to use Python with Blink1Control running, you can issue REST HTTP requests to Blink1Control's built in REST API server.

andkon, 2017-10-17:
I believe I didn't have Blink1Control installed when trying to run it. Let me give it a shot again though. Is there documentation handy for Blink1Control's REST endpoints?

todbot, 2017-10-17:
yup.
https://github.com/todbot/blink1/blob/master/docs/app-url-api-examples.md
https://github.com/todbot/blink1/blob/master/docs/app-url-api.md

todbot, 2017-10-18:
I have also verified that the `blink1` library on pypi (and in here) exhibits the `USBError: [Errno 13] Access denied (insufficient permissions)` on Mac OS X 10.12.6.  Looking into what the problem could be, but I don't know much about this Python module (it was written by another dev)

andkon, 2017-10-18:
Cool, I'll keep following this thread, thank you for digging in! From work I know how hard it is to manage+maintain a ton of different platforms and bindings, so I really do appreciate it.

DavyK, 2017-11-30:
I am also experiencing this issue

andkon, 2017-12-05:
Still having this on High Sierra 10.13.1 :(

todbot, 2017-12-05:
This inability of the Python blink(1) library to access devices is because of the PyUSB library. Both Mac & Windows will have issues as they both have built-in HID drivers that take over USB HID devices (and provide nice abstractions to make access HID devices easier). So, the blink(1) library needs to be rewritten to use `hid` Python library. 

In the meantime, you could try one of the alternative libraries here:
https://github.com/todbot/blink1/tree/master/python/alternative_libraries